ARMmbed/greentea #23: Support for J-Link to flash target

KaveriBanakar, 2015-10-15:
Hi,
MbedOS porting on new microcontroller is in progress. Currently I can use only J-Link to flash binaries onto my board. But mbedhtrun expects below copy methods to flash binary:
-c COPY_METHOD, --copy=COPY_METHOD
                      Copy (flash the target) method selector. Plugin
                      support: copy, cp, default, eACommander, eACommander-
                      usb, shell, shutil, stlink, xcopy
Is there a way to use J-Link to flash binary in mbedhtrun command?

rainierwolfcastle, 2015-10-15:
http://jira-dev0.cambridge.arm.com/browse/IOTSYST-192

rainierwolfcastle, 2015-10-16:
ARM Internal Ref: IOTSYST-452

PrzemekWirkus, 2015-10-19:
@KaveriBanakar Hi, currently there is no support for J-link for htrun commands.
Can you give me an example how you flash devices with J-Link, we will think about how to add support for J-LInk to mbedhtrun command line?

0xc0170, 2015-10-19:
@PrzemekWirkus check this tutoria: http://community.silabs.com/t5/32-bit-MCU-Knowledge-Base/Using-JLink-Commander-to-program-flash/ta-p/133136 , how to use jlink command line. The basic covered there is to set target, erase, flash binary and restart mcu

PrzemekWirkus, 2015-10-21:
@0xc0170 Thanks! I will have a look.

PrzemekWirkus, 2015-11-25:
> Is there a way to use J-Link to flash binary in mbedhtrun command? 

I think for now we (as mbed team) will not add support for j-link.
You can always contribute if you want. We will definitely do it in the near future though!

HamedSiasi, 2016-05-11:
Hi,
any update about supporting j-link ?
Thanks

Eosis, 2017-11-13:
We would also be interested in using JLink to flash tests while using GreenTea.

Is there any guidance as to where to start adding this functionality in if we wanted to contribute?

theotherjimmy, 2017-11-13:
Hi @Eosis,

It's great to hear that you want to contribute. I'm pretty sure that flashing is handled by [armmbed/htrun](https://github.com/armmbed/htrun).

You would probably add a new plugin into the [plugins folder](https://github.com/ARMmbed/htrun/tree/master/mbed_host_tests/host_tests_plugins) and register your plugin in the [block of plugin registrations](https://github.com/ARMmbed/htrun/blob/master/mbed_host_tests/host_tests_plugins/__init__.py#L51-L72).

@bridadan Does that look correct to you?

bridadan, 2017-11-13:
@theotherjimmy yup that's correct!

closed 2018-10-02 by adbridge