ARMmbed/greentea #25: Build files(.bin) exist in test folder, test framework can't load it in frdm-k64f

hasnainvirk, 2015-10-26:
Setup: Linux (VM-ware)
Board: frdm-k64f
compiler: gcc & armcc
K64f firmware versions: 0226 and 0225
ISSUE with GCC: with GCC all tests pass with status OK. However A message appears.
cp: cannot create regular file ‘/media/hasnain/MBED/XXXX.bin’: No such file or directory
ISSUE with armcc: same as gcc alongwith some test cases Timinig out.

TRACE for gcc:

hasnain@ubuntu:~/Desktop/OOB/mbed-drivers$ sudo mbedgt --target=frdm-k64f-gcc
mbedgt: detecting connected mbed-enabled devices... 
mbedgt: detected 1 device
	detected 'K64F', console at '/dev/ttyACM0', mounted at '/media/hasnain/MBED'
mbedgt: scan available targets for 'K64F' platform...
mbedgt: yotta search for mbed-target 'k64f'
	calling yotta: yotta search -k mbed-target:k64f target
mbedgt: using 'frdm-k64f-gcc' target, prepare to build
mbedgt: building your sources and tests with yotta...
	calling yotta: yotta -v --target=frdm-k64f-gcc,* build
info: generate for target: frdm-k64f-gcc 0.2.0 at /home/hasnain/Desktop/OOB/mbed-drivers/yotta_targets/frdm-k64f-gcc
GCC version is: 4.9.3
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hasnain/Desktop/OOB/mbed-drivers/build/frdm-k64f-gcc
ninja: no work to do.
mbedgt: yotta build for target 'frdm-k64f-gcc' was successful
mbedgt: running tests for target 'frdm-k64f-gcc'
	running host test...
	test 'mbed-drivers-test-serial_interrupt' .............................................. OK in 6.31 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-blinky.bin’: No such file or directory
	test 'mbed-drivers-test-blinky' ........................................................ OK in 6.23 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-div.bin’: No such file or directory
	test 'mbed-drivers-test-div' ........................................................... OK in 6.26 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-cstring.bin’: No such file or directory
	test 'mbed-drivers-test-cstring' ....................................................... OK in 6.21 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-stdio.bin’: No such file or directory
	test 'mbed-drivers-test-stdio' ......................................................... OK in 6.21 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-stl.bin’: No such file or directory
	test 'mbed-drivers-test-stl' ........................................................... OK in 6.30 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-rtc.bin’: No such file or directory
	test 'mbed-drivers-test-rtc' ........................................................... OK in 6.28 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-dev_null.bin’: No such file or directory
	test 'mbed-drivers-test-dev_null' ...................................................... OK in 6.22 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-cpp.bin’: No such file or directory
	test 'mbed-drivers-test-cpp' ........................................................... OK in 6.23 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-timeout.bin’: No such file or directory
	test 'mbed-drivers-test-timeout' ....................................................... OK in 6.20 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-basic.bin’: No such file or directory
	test 'mbed-drivers-test-basic' ......................................................... OK in 6.17 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-ticker.bin’: No such file or directory
	test 'mbed-drivers-test-ticker' ........................................................ OK in 6.08 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-ticker_2.bin’: No such file or directory
	test 'mbed-drivers-test-ticker_2' ...................................................... OK in 6.21 sec
	running host test...
cp: cannot create regular file ‘/media/hasnain/MBED/mbed-drivers-test-heap_and_stack.bin’: No such file or directory
	test 'mbed-drivers-test-heap_and_stack' ................................................ OK in 6.19 sec
	running host test...
[... 136 lines cut ...]
+-----------------+---------------+------------------------------------+---------+--------------------+-------------+
| frdm-k64f-armcc | K64F          | mbed-drivers-test-serial_interrupt | TIMEOUT | 20.26              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-blinky           | TIMEOUT | 20.26              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-div              | TIMEOUT | 20.26              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-dev_null         | TIMEOUT | 20.26              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-stdio            | OK      | 6.24               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-sleep_timeout    | TIMEOUT | 20.27              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-ticker           | OK      | 6.29               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-rtc              | OK      | 6.24               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-cstring          | TIMEOUT | 20.26              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-cpp              | OK      | 6.25               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-timeout          | TIMEOUT | 20.25              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-basic            | OK      | 6.19               | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-ticker_3         | TIMEOUT | 20.27              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-ticker_2         | TIMEOUT | 20.27              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-heap_and_stack   | TIMEOUT | 20.26              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-hello            | TIMEOUT | 20.26              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-time_us          | TIMEOUT | 20.27              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-stl              | TIMEOUT | 20.26              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-echo             | TIMEOUT | 20.27              | shell       |
| frdm-k64f-armcc | K64F          | mbed-drivers-test-detect           | TIMEOUT | 20.27              | shell       |
+-----------------+---------------+------------------------------------+---------+--------------------+-------------+

Result: 5 OK / 15 TIMEOUT
Completed in 580.94 sec

simonqhughes, 2015-10-26:
ARM Internal Ref: IOTSYST-478

PrzemekWirkus, 2015-11-05:
We still do not support VM-ware Linux distros.

ErkkiMehtala, 2015-11-10:
Should it be then mentioned in readme 'Supported operating systems' so no-one wastes time on VM for this?

0xc0170, 2015-11-10:
We got a list of known issues, Przemek is this there? @ErkkiMehtala I would say yes. Przemek, what do you think?

closed 2016-03-11 by PrzemekWirkus