ARMmbed/greentea #26: Win7: 'NoneType' object has no attribue 'lower'

niklas-arm, 2015-10-27:
OS: Windows 7 SP1 inside VirtualBox on OS X host

Installation: After installing yotta using the Windows installer v0.1.4 and then installing greentea using `pip install mbed-greentea` inside the yotta command shell.

Setup: Using `mbed-drivers` as a test environment with `frdm-k64f-gcc` target and a K64F board connected to the VM. `mbedls` output:
```
(workspace) C:\Users\username\mbed-drivers>mbedls
+---------------------+----------------------------+-------------------+-------------------+--------------------------------+
|platform_name        |platform_name_unique        |mount_point        |serial_port        |target_id                       |
+---------------------+----------------------------+-------------------+-------------------+--------------------------------+
|K64F                 |K64F[0]                     |F:                 |COM4               |02400201AA3F4E7757C1B3CF        |
|unknown              |None[0]                     |E:                 |COM3               |066BFF555055835087015049        |
+---------------------+----------------------------+-------------------+-------------------+--------------------------------+
```
I'm not sure what the `unknown` platform name is.

Problem: Python throws a stack-trace when attempting `mbedgt -t frdm-k64f-gcc`
```
(workspace) C:\Users\username\mbed-drivers>mbedgt -t frdm-k64f-gcc
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 2 devices
        detected '', console at 'COM3', mounted at 'E:'
mbedgt: scan available targets for '' platform...
mbedgt: Unexpected error:
        'NoneType' object has no attribute 'lower'
Traceback (most recent call last):
  File "C:\yotta\python\Lib\runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "C:\yotta\python\Lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\yotta\workspace\Scripts\mbedgt.exe\__main__.py", line 9, in <module>
  File "c:\yotta\workspace\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 210, in main
    cli_ret = main_cli(opts, args)
  File "c:\yotta\workspace\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 318, in main_cli
    mut_info = get_mbed_clasic_target_info(mut['platform_name'])
  File "c:\yotta\workspace\lib\site-packages\mbed_greentea\mbed_target_info.py", line 187, in get_mbed_clasic_target_info
    TARGET_INFO_MAPPING = add_target_info_mapping(mbed_classic_name)
  File "c:\yotta\workspace\lib\site-packages\mbed_greentea\mbed_target_info.py", line 145, in add_target_info_mapping
    yotta_target_search = get_mbed_targets_from_yotta(mbed_classic_name)
  File "c:\yotta\workspace\lib\site-packages\mbed_greentea\mbed_target_info.py", line 117, in get_mbed_targets_from_yotta
    cmd = ['yotta', '--plain', 'search', '-k', 'mbed-target:%s'% mbed_classic_name.lower().strip(), 'target']
AttributeError: 'NoneType' object has no attribute 'lower'
```

PrzemekWirkus, 2015-10-28:
Thanks for reporting!

rainierwolfcastle, 2015-10-28:
ARM Internal Ref: IOTSYST-488

PrzemekWirkus, 2015-11-03:
Fixed with [latest release](https://github.com/ARMmbed/greentea/releases/tag/v0.1.8).

closed 2015-11-03 by PrzemekWirkus