ARMmbed/greentea #32: Please add support for coverage output

bremoran, 2015-11-19:
This implements the host side of gcov code coverage on mbed-gcc.  The device side is implemented in:

* https://github.com/ARMmbed/mbed-drivers/pull/119
* https://github.com/ARMmbed/target-mbed-gcc/pull/19

When a test completes, ```MBED_HOSTTEST_RESULT()``` emits either ```{{success}}``` or ```{{failure}}``` and ```{{end}}``` as normal.  Following ```{{end}}```, ```gcov_exit()``` is invoked.  gcov expects to be operating on a filesystem, where the code was compiled, so it attempts to output to gcda files in the same directory as the object file that they cover. ```_open()``` is overridden to output the file name in a greentea-style output tag. _write is overridden to output hex-encoded bytes, and close is overridden to encode another greentea-style output tag for example:

```
{{timeout;20}}
{{host_test_name;default_auto}}
{{description;Basic}}
{{test_id;MBED_A1}}
{{start}}
{{success}}
{{end}}
{{coverage_start;path/to/build/frdm-k64f-gcc/ym/ualloc/source/CMakeFiles/ualloc.dir/malloc.c.gcda}}
616463677238303436e9841b000000a370000000743e2891d40d000001000000b67a050000000000
1fae0000000000001fae000000000000ff3fc5000000000200000000000000000000000000000000
00000000000000005f0b000000000000000000000000000000000000800100000100000000000000
80010000000000009700000002000000000000002e01000000000000060000000300000000000000
12000000000000000800000004000000000000002000000000000000050000000500000000000000
19000000000000000800000006000000000000003000000000000000090000000700000000000000
3f000000000000000d00000008000000000000006900000000000000040000000b00000000000000
2c000000000000000a0000000c000000000000007c00000000000000030000000e00000000000000
2c000000000000000400000010000000000000004000000000000000050000001500000000000000
69000000000000000100000020000000000000002000000000000000010000003200000000000000
32000000000000000100000066000000000000006600000000000000080000007700000000000000
b803000000000000080000001fae000000000000f870050000000000000000010300000005000000
9d555c453eb2bbc00000a10104000000000000000000000000000000000000000000000103000000
040000001f1780513eb2bbc00000a101040000000000000000000000000000000000000000000001
03000000030000005bd527c2f94ab37c0000a1010400000000000000000000000000000000000000
000000010300000002000000a37cb51df94ab37c0000a10104000000000000000000000000000000
000000000000000103000000010000006d9f0213f94ab37c0000a101040000000500000000000000
050000000000000000000000{{coverage_end}}
```

Ideally, greentea would read this output and create a file at ```path/to/build/frdm-k64f-gcc/ym/ualloc/source/CMakeFiles/ualloc.dir/malloc.c.gcda``` and populate it with the binary equivalent of everything between the coverage tags.  The ```{{coverage_start;<path>}}``` and ```{{coverage_end}}``` tags can occur many times, one of each file in the build.

Currently, only gcov output is supported, since it can be generated without a trace port.

One possible optimization would be to add a command to change the serial port speed when the coverage output starts.  Coverage of even mbed-test-basic generates enormous quantities of data.

cc @PrzemekWirkus

PrzemekWirkus, 2015-11-19:
I will check if having this after {{end}} will be OK because we assume {{end}} is really end ;)
So let me check that and I will get back to you today or tomorrow.

bremoran, 2015-11-19:
I have moved ```{{end}}``` to after the coverage output.

PrzemekWirkus, 2015-11-20:
Let's agree on a format for multiple coverage data blocks in one test run:

Each coverage block should have 2 lines:
* Line 1: ```{{coverage_start;PATH}}``` where PATH is path to payload destination file
* Line 2: ```PAYLOAD{{coverage_end}}``` where PAYLOAD is hex ascii encoded binary data.

## Coverage block example
```
{{coverage_start;/path/to/lcov_file.c.gcda}}
616463677238303436e9841b000000a370000000743e2891d40d0000{{coverage_end}}
```
Where:
* PATH to payload file: ```/path/to/lcov_file.c.gcda```
* PAYLOAD ```616463677238303436e9841b000000a370000000743e2891d40d0000```

## Multiple coverage block example
```
{{coverage_start;/path/to/lcov_file_1.c.gcda}}
616463677238303436e9841b000000a370000000743e2891d40d0000{{coverage_end}}
{{coverage_start;/path/to/lcov_file_2.c.gcda}}
1000000b67a0500000000001fae0000000000001fae000000000000f{{coverage_end}}
```
* Note: Corresponding greentea log
```
mbedgt: storing coverage data artefacts
        storing 28 bytes in '/path/to/lcov_file_1.c.gcda'
        storing 28 bytes in '/path/to/lcov_file_2.c.gcda'
```

bremoran, 2015-11-20:
That looks good to me, except that I thought we were adding the data format:

```
{{coverage_start;hex;/path/to/lcov_file_1.c.gcda}}
616463677238303436e9841b000000a370000000743e2891d40d0000{{coverage_end}}
{{coverage_start;hex;/path/to/lcov_file_2.c.gcda}}
1000000b67a0500000000001fae0000000000001fae000000000000f{{coverage_end}}
```

PrzemekWirkus, 2015-11-20:
We can add data format at this stage.
I've created https://github.com/ARMmbed/greentea/pull/34 to initially start working on this (no data format support there).
You can locally clone this PR and do ```python setup.py install``` to see if coverage is stored properly with your changes.

bremoran, 2015-11-20:
It might be a good idea to add a checksum too.

PrzemekWirkus, 2015-11-23:
Good idea, I propose to add it to ```{{coverage_end}}```.

Example:
```{{coverage_end;CRC32}}``` where CRC32 could be asciii hex string with CRC32 checksum.

PrzemekWirkus, 2015-11-23:
We got few things to solve here:
* Flashing timeout now is 20 seconds, should be longer.
* Test case timeout should not include ```{{coverage_start}}``` ... ```{{coverage_end}}```

PrzemekWirkus, 2015-12-07:
Latest merged for this in [mbed-drivers](https://github.com/ARMmbed/mbed-drivers/pull/119).

PrzemekWirkus, 2016-03-04:
Added in version 0.2.0.

closed 2016-03-04 by PrzemekWirkus