ARMmbed/greentea #35: Post-test hook

bremoran, 2015-11-23:
When generating coverage output, a command needs to be run after each test to aggregate the data that was generated from the test. This should not be specific to ```gcov```, so a generic post-test hook seems like the right solution.

This feature is to support #32.

For the specific gcov use-case, after each test, the following commands need to be run:
```
lcov --gcov-tool arm-none-eabi-gcov  --capture --directory <build path> --output-file <build path>/test/<test name>.info
find . -name "*.gcda" | xargs rm -f
```

After all tests have run, the following command should be run, one ```-a``` entry for each passing test:
```
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test1 name>.info [-a <build path>/test/<test2 name>.info] --output-file <result path>.info
```

This will aggregate all data generated from each test into its own ```lcov``` info file. These info files can then be aggregated into a single info file covering all tests that have run.

The post-test hook could be either a python script or a command line. If a command line, then accepting a command to be parsed with the python format language might be beneficial, for example:

```Python
mbedgt --post-test-hook="lcov --gcov-tool arm-none-eabi-gcov  --capture --directory {{build_path}} --output-file {{build_path}}/test/{{test_name}}.info"
```

PrzemekWirkus, 2015-12-14:
Brendan,
This command (after all tests are completed)
```
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test1 name>.info \
[-a <build path>/test/<test2 name>.info] --output-file <result path>.info
```
should expand to:
* a series of commands or 
```
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test_1>.info --output-file <result path>.info
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test_2>.info --output-file <result path>.info
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test_3>.info --output-file <result path>.info
...
```
* "one liner" command line:
```
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test_1>.info -a <build path>/test/<test_2>.info -a <build path>/test/<test_3>.info --output-file <result path>.info
```
?

PrzemekWirkus, 2015-12-14:
This issue will be addressed by this change https://github.com/ARMmbed/greentea/pull/34.

bremoran, 2015-12-14:
A series of commands will work:

```
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test_1>.info -a <result path>.info --output-file <result path>.info
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test_2>.info -a <result path>.info --output-file <result path>.info
lcov --gcov-tool arm-none-eabi-gcov -a <build path>/test/<test_3>.info -a <result path>.info --output-file <result path>.info
```

Note that there is a difference from the above. To aggregate results, lcov needs to load the intermediate file as well as the result file.

Otherwise, the oneliner is fine.

bremoran, 2015-12-14:
Note that by expanding the lcov commands to a series of commands, we can use the framework for calling one command per test to do both the gcov and lcov aggregation commands.

PrzemekWirkus, 2016-03-04:
Added in version 0.2.0.

closed 2016-03-04 by PrzemekWirkus