ARMmbed/greentea #39: Quickstart - CppUtest references using private module

0xc0170, 2015-12-01:
File: https://github.com/ARMmbed/greentea/blob/master/docs/QUICKSTART.md

As this is a private module, it's not available thus shall we remove the references for now? There will be replaced or point to a new module once it's ready.

PrzemekWirkus, 2015-12-01:
You can send PR! :)

brianjang, 2015-12-02:
then you mean, for mbed OS 3.0 can't use cppUTest framework?
- as far as i know, mbed OS 3.0 has greentea test-framework.(ex, for testing mbed-drivers)
Could you give some information the below part?
-          How can I get cppUtest framework for mbed OS 3.0?
-          How can I build cppUtest test code in “yotta_modules\mbed-client-c\test”?
(from mbed github, mbed-client-c include "test" and "unittest" folder and these test code of these folder is using cppUTest framworks API)

PrzemekWirkus, 2015-12-08:
@brianjang 
I think for general docs you need to use: https://github.com/ARMmbed/greentea/tree/master/docs
Just a note of clarification:
* Greentea is a program used to automate test execution on mbed devices. Basically it builds yotta module and tests inside and flashes all tests one by one to get test results.
* CppUTest is a test framework, C/C++ code used to build together with your test code e.g. to mock, perform assertions etc.
  * You can compile your modules with [```testDependency```](http://yottadocs.mbed.com/reference/module.html#testDependencies) to CppUTest module (not sure if published our module with CppUTest) 
  * [Example test dependencies](https://github.com/ARMmbed/core-util/blob/master/module.json#L24) for this yotta module can be updated with any yotta module with additional test frameworks like [CppUTest](http://cpputest.github.io) or [Unity](https://github.com/ThrowTheSwitch/Unity). As long it compiles! ;)

BTW:
>How can I build cppUtest test code in “yotta_modules\mbed-client-c\test”? (from mbed github, mbed-client-c include "test" and "unittest" folder and these test code of these folder is using cppUTest framworks API)

I think they are using Unity there with ```mbed-client-c```. This solution is non canonical and you should file an Issue there (mbed-client-c) to ask them how to build stuff they have prepared with their module :)

niklas-arm, 2015-12-18:
@brianjang Have a look at our recommended test harness [`utest`](https://github.com/ARMmbed/utest) which works with the [UNITY test macros](https://github.com/ARMmbed/unity) and integrates with greentea.

closed 2016-01-12 by PrzemekWirkus