ARMmbed/greentea #44: greentea fails to run with a local host-test

bremoran, 2015-12-17:
Here is my output
```
$ mbedgt -V --target frdm-k64f -n $testname
Traceback (most recent call last):
  File "/usr/local/bin/mbedgt", line 9, in <module>
    load_entry_point('mbed-greentea==0.1.14', 'console_scripts', 'mbedgt')()
  File "/usr/local/lib/python2.7/site-packages/mbed_greentea/mbed_greentea_cli.py", line 105, in main
    copy_methods_str = "Plugin support: " + ', '.join(mbed_host_tests.host_tests_plugins.get_plugin_caps('CopyMethod'))
NameError: global name 'mbed_host_tests' is not defined
```

I have tried removing and reinstalling mbed-greentea, mbed-host-tests, and mbed-ls

bremoran, 2015-12-17:
The cause appears to be this:

```
import mbed_host_tests
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/mbed_host_tests/__init__.py", line 40, in <module>
    from host_tests import BaseHostTest
  File "/usr/local/lib/python2.7/site-packages/mbed_host_tests/host_tests/__init__.py", line 35, in <module>
    from mbed_client_auto import LWM2MClientAutoTest
  File "/usr/local/lib/python2.7/site-packages/mbed_host_tests/host_tests/mbed_client_auto.py", line 32, in <module>
    class TestConfiguration():
  File "/usr/local/lib/python2.7/site-packages/mbed_host_tests/host_tests/mbed_client_auto.py", line 41, in TestConfiguration
    OWN_PC_ADDRESS = socket.gethostbyname(socket.gethostname())
socket.gaierror: [Errno 8] nodename nor servname provided, or not known

```

PrzemekWirkus, 2016-01-04:
This is because mbed_client_auto-.py host test is naughty - it will be removed with this https://github.com/ARMmbed/htrun/pull/43.

PrzemekWirkus, 2016-01-04:
I will close it when I merge and release new [htrun](https://github.com/ARMmbed/htrun).

PrzemekWirkus, 2016-01-08:
Change was merged in htrun, I will close this issue.

closed 2016-01-08 by PrzemekWirkus