ARMmbed/greentea #54: It does not release the serial port when timeouts

0xc0170, 2016-01-06:
It's happening lately that I have to restart a board. I could not find this reported (open or close), I recall it was already reported.

Env: 
Target: k64f, OS: windows, mbedgt 0.1.14

```
// mbedgt time outs because of the test block
// running mbedgt again
mbedgt: mbed-host-test-runner: started
MBED: Instrumentation: "COM64" and disk: "D:"
HOST: Copy image onto target...
        1 file(s) copied.
HOST: Initialize serial port...
........................................mbedgt: mbed-host-test-runner: stopped
mbedgt: mbed-host-test-runner: returned 'TIMEOUT'
```

PrzemekWirkus, 2016-01-07:
Good, I will investigate this ASAP.

mazimkhan, 2016-01-08:
@PrzemekWirkus  this issue may be related to issue #48 
restarting the target may be helping mbed host test to exit as well. Anyways closing the serial port should be part of mbed host test when mbed gt times out

PrzemekWirkus, 2016-01-12:
Yes, let's see how this issues behaves when #48 is fixed.

mazimkhan, 2016-01-14:
It should be fixed by:
ARMmbed/htrun#54
#63

closed 2016-01-20 by mazimkhan