ARMmbed/greentea #55: Host test error - hidden, does not stop the execution

0xc0170, 2016-01-07:
The console is misleading and does not highlight the error, still using 0.1.14 mbedgt.

```
HOST: Error! While loading local host test module 'C:\Code\git_repo\github\yotta_thing\niklas-socket-test\test\host_tests\tcp_echo_client.py
'                                                                                                                                           
HOST: invalid syntax (tcp_echo_client.py, line 90)                                                                                          
MBED: Instrumentation: "COM64" and disk: "D:"                                                                                               
HOST: Copy image onto target...                                                                                                             
        1 file(s) copied.                                                                                                                   
HOST: Initialize serial port...                                                                                                             
...port ready!                                                                                                                              
HOST: Reset target...                                                                                                                       
HOST: Detecting test case properties...                                                                                                     
HOST: Property 'timeout' = '60'                                                                                                             
HOST: Property 'host_test_name' = 'tcp_echo_client'                                                                                         
HOST: Property 'description' = 'TCP echo client unit test'                                                                                  
HOST: Property 'test_id' = 'TCP echo client unit test'                                                                                      
HOST: Start test...                                                                                                                         
HOST: Error! Unknown host test name 'tcp_echo_client' (use 'mbedhtrun --list' to verify)!                                                   
HOST: Error! You can use switch '-e <dir>' to specify local directory with host tests to load                                               
```

I assume the error:
HOST: Error! While loading local host test module 'C:\Code\git_repo\github\yotta_thing\niklas-socket-test\test\host_tests\tcp_echo_client.py

should stop the execution ?

0xc0170, 2016-01-07:
I had syntax error in the host test, I discovered this when I run the script manually. Thus to reproduce this, create a host test with a syntax error.

mazimkhan, 2016-01-07:
Analysis:

The errors "HOST: Error! While loading local ho...."   and "HOST: Error! Unknown host test name 'tcp_echo_client'...." originate from different places. 

Because of syntax error in the host test script tcp_echo_client.py it is skipped. mbedhtrun continues as there are other host test scripts that load fine.

When the test binary is loaded it asks for host test "'host_test_name' = 'tcp_echo_client'"  that failed to load. Hence mbed ht run gives error "Unknown host test". 

The binding between the host test script and a binary is the property printed on the serial port like "Property 'host_test_name' = 'tcp_echo_client'". 
Continuing after one faulty host test script gives a bit of fault tolerance. We have to decide whether we just fail one test because of this or entire test suite.

mazimkhan, 2016-01-07:
Just so that you understand how mbed host test runs: 
1. detects all host tests. Ignores erroneous ones.
2. detects all binaries
3. for each binary: flashes it and runs host test
   3.1 binary prints out which host test module/script it needs.   Correspondence to the host test happens here
   3.2 start the requested host test script

From your comment it seems your understanding is that a flashing and testing happens for each host test. And that is wrongly happening for a faulty host test script.    Flashing and testing happens for each test binary. Some binaries may refer to a same host test.

0xc0170, 2016-01-08:
I should have provided a command in the code snippet, I run only one test using ``` mbedgt -V -n socket-test-test-tcp_client_echo_v0 ```

Thanks for those details. if htrun captures in my case 'tcp_echo_client' failure, it should check if that host test failed to load (``` HOST: invalid syntax (tcp_echo_client.py, line 90)    ```), we could improve the message: ``` HOST: Error! Unknown host test name 'tcp_echo_client' (use 'mbedhtrun --list' to verify)!  ``` to  ``` HOST: 'tcp_echo_client' host test failed to load, reason : reason_here  ```?

It's not bug, enhancement

mazimkhan, 2016-01-08:
As I said before until the binary is read and host test module is read on the serial port we don't which script OR module in the script it requires. Just based on the faulty script name we can not assume that a binary will refer it by that name. Hence there is no concrete information to device a message like 
HOST: 'tcp_echo_client' host test failed to load, reason : reason_here?   --- it can be wrong

But I will update the documentation in mbed-host-test repo and mention this example.

mazimkhan, 2016-01-08:
Created issue in mbed host test for updating documentation:
https://github.com/ARMmbed/htrun/issues/51

closed 2016-01-08 by mazimkhan