ARMmbed/greentea #60: Improve ambiguous missing serial port error message

Patater, 2016-01-12:
Currently, when greentea can't find the serial port, it outputs an ambiguous error message that doesn't indicate to the user what went wrong or how to fix the problem.

```shell
$ mbedgt -V 
mbedgt: checking for yotta target in current directory
        reason: no --target switch set
mbedgt: checking yotta target in current directory
        calling yotta: yotta --plain target
mbedgt: assuming default target as 'frdm-k64f-gcc'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: can't detect all properties of the device!
        use 0 instances for testing
mbedgt: no platform/target matching tests were found!
mbedgt: no target matching platforms were found!
mbedgt: exited with code -110
mbedgt: completed in 0.39 sec
$ mbedls
+---------------------+----------------------------+---------------------+-------------------+--------------------------------+
|platform_name        |platform_name_unique        |mount_point          |serial_port        |target_id                       |
+---------------------+----------------------------+---------------------+-------------------+--------------------------------+
|K64F                 |K64F[0]                     |/Volumes/MBED        |unknown            |0240020152A06E54AF5E93EC        |
+---------------------+----------------------------+---------------------+-------------------+--------------------------------+
```

This error message should be improved to indicate to the user what went wrong exactly: that greentea can't find the serial port for the device.

PrzemekWirkus, 2016-01-12:
Will go with this PR: https://github.com/ARMmbed/greentea/pull/62

Patater, 2016-01-12:
Excellent. Thanks.

closed 2016-01-15 by PrzemekWirkus