ARMmbed/greentea #66: Add support for test case id tagging

PrzemekWirkus, 2016-01-15:
One binary can contain may test cases (test fixtures) with a unique test case ids.
We could add support to yield many test case id results per one test case execution.

PrzemekWirkus, 2016-01-15:
DoD:
* Documentation,
* Macro on test case side (C/C++),
* Support for greentea and macros for test case ids.

0xc0170, 2016-01-18:
I don't understand this feature, please can you provide C/C++ and python side code snippets.

Provided code snippet below does not make much sense as it is now to me - BaseHostTest.test method is blocking at the moment. Is this a step forward for async BaseHostTest.test ? Why do we need a name parameter? why selftest does not print the current running one? How do I assign the testaname ? Random number generator ?
```
class RTCTest(BaseHostTest):
    def test(self, selftest):
        test_result = True
        selftest.testcase_start("RTC0001")
        # Test case execution
        selftest.testcase_finish("RTC0001", test_result)
        # ...
```

Why testcase_start is not automatic (once test() is invoked), case start is indicated (it might be useful to have testcase_start manual method?), and testcase finish - does it need 2 arguments, why the selftest or self does not contain result , plus test id is automatic?

PrzemekWirkus, 2016-01-18:
@0xc0170 Martin, I will provide and example, this description is not finished yet. I need to provide a guide and examples to fully cover this feature. I just had no time to finish this yesterday.

PrzemekWirkus, 2016-01-18:
Reviewed this and other proposals today with Brenda, Martin and Niklas.

0xc0170, 2016-01-19:
> MBED_HOSTTEST_TESTCASE_FINISH(TESTCASE_NAME,success);

we use result, not success? I saw in host tests results, and ``` MBED_HOSTTEST_RESULT(RESULT)  
```

From the client side (c/c++) as it is (2 new macros) is fine, but from the host side, how htrun knows how many test_ are there, is it like with test frameworks, that a method needs to start with test_ ?
```
def test_rtc001(self, selftest):
  # do magic

def test_rtc100(self, selftest):
  # do magic
```

```
selftest.testcase_start("RTC0002")
```

Wouldn't automatic naming make sense (test_ name), name would be upper case -> RTC001, RTC100 in my case. Or just a variable to set (same as for BaseHosttest, we are setting name, not calling a method), selftest.testcase.name = RTC001 (this is the name of the current test case, it starts once method test_rtc001 is invoked, finished once returned). However, if the goal for this testcases is to be async in the future, then we might rethink this concept. Then it might be better to have TestCase class which would provide start(), finish(), set_timeout() ?

PrzemekWirkus, 2016-01-19:
* ```MBED_HOSTTEST_RESULT(RESULT)``` macro is used to return result from test suite (binary with all tests).
* ```MBED_HOSTTEST_TESTCASE_FINISH(TESTCASE_NAME,SUCCESS);``` macro is used to return 'success' from each test case inside test suite.

So your pseudo-code for one binary would look like:
```c++
int main() {

    {
        MBED_HOSTTEST_TESTCASE_START(TESTCASE_NAME);
        # test case code
        MBED_HOSTTEST_TESTCASE_FINISH(TESTCASE_NAME,SUCCESS);
    }
    {
        MBED_HOSTTEST_TESTCASE_START(TESTCASE_NAME);
        # test case code
        MBED_HOSTTEST_TESTCASE_FINISH(TESTCASE_NAME,SUCCESS);
    }
    ...
    {
        MBED_HOSTTEST_TESTCASE_START(TESTCASE_NAME);
        # test case code
        MBED_HOSTTEST_TESTCASE_FINISH(TESTCASE_NAME,SUCCESS);
    }

    MBED_HOSTTEST_RESULT(RESULT);
    return 0;
}
```
In above code you have three test cases execution and at the end you return test suite result.

PrzemekWirkus, 2016-01-19:
I can imagine ```MBED_HOSTTEST_RESULT(RESULT)``` completely disheartening after we refactor whole macro and sync stuff. You probably remember our yesterday talk with Niklas and Brendan, we gonna issue ton of tickets to improve a lot this things.
But this is a step towards a good direction and backward compatibility with older tests.

PrzemekWirkus, 2016-01-19:
> Wouldn't automatic naming make sense (test_ name), name would be upper case -> RTC001, RTC100 in my case. 
Yes it would, we discussed this yesterday with N. & B. You probably missed it :)

mazimkhan, 2016-01-19:
Comment on grouping of tests. The example mbedgt test report gives idea:
1. That ```rtc``` and ```cstring``` binaries will have subtests. Is it true.
2. Does it also mean that <repo>/test/<test name>/main.cpp will have sub tests?
3. Is the one binary multiple tests solution only for subtests?

My understanding of one binary having multiple tests was that one binary can have all or a group of tests based on the target memory size? and it will use clitest interface for test indexing, start and verdict.

niklas-arm, 2016-01-19:
I wholeheartedly disagree with returning `success` as a signed integer!
Returning `404`, or `-42` as an integer does not make sense, when you can use printf to directly output a failure string.
What is the python side supposed to do with an integer? Will you first print a string that maps the error code to a failure string? This makes no sense.

What is the semantical difference between Error and Fail? Both are "not success" and therefore the test didn't pass.
Perhaps Error and Warning, where Error prints and fails, and Warning prints, but continues, might be a better scheme.
Then again: This is a (unit-) test, why would you have warning in there? And why would you ignore them?

Error codes are used because cheap to pass around (hence an int) and machine readable.
We don't need that. We need human readable, understandable error messages.
Why not just be verbose and print why it failed?

niklas-arm, 2016-01-19:
`selftest.testcase_start("RTC0001")`
vs.
`MBED_HOSTTEST_TESTCASE_START(STL002);`

IT'S A ~~TRAP~~ STRING.
`MBED_HOSTTEST_TESTCASE_START("STL002");`

There is absolutely no reason to not use a string here. None.

mazimkhan, 2016-01-19:
@niklas-arm regarding difference between Error and Fail. We need this to differentiate between between actual test failure and automation failure. Example the target wanted an input from host test side to carry on the test and failed on timeout. So not really a test failure.

I more inclined towards using word INCONCLUSIVE. This status would be like an error for for automation/system test team. Device software team only need to debug Failures.

However, I am not sure we should restrict detection of automation issues on host side only. So only PASS and FAIL from target?

mazimkhan, 2016-01-19:
Also I advocate only three top level verdicts INCONC/ERROR (inconclusive/automation error), FAIL and PASS. While communicating INCONC/ERROR or FAIL the test can option pass relevent information in form of a string or error code that easily understandable in the report.

mazimkhan, 2016-01-19:
```test_rtc001``` is a subtest proposal on host side. Do we need this? What if two sub tests in two different main tests need same logic. If a binary can print a list of tests coded in it and their corresponding host tests, mbed-host-test can load the required host test in the same fashion as it does now. Hence I don't see reason to implement this.

PrzemekWirkus, 2016-01-19:
@mazimkhan We were till now able to conclude test (case) execution result on DUT and/or host test side.
* DUT side - binary can conclude test case result itself and print {success}{end}
* DUT or host test - For example in network test both sides can say "error" / "fail" - e.g. conenction can be dropped on one or the other side.
* Host test - host test decides if test (case) passes / fails / errors - e.g. timer tests: DUT can't measure time on itself.

niklas-arm, 2016-01-19:
@mazimkhan Ok, good point. I would be fine with an "automation" error, which signifies failures related to running the test, not the test itself.

PrzemekWirkus, 2016-03-04:
Added in version 0.2.0.

closed 2016-03-04 by PrzemekWirkus