ARMmbed/greentea #68: Feature request: Sort tests in order in list and report

LiyouZhou, 2016-01-18:
When you have many test to run, it is difficult to find the test you want in the list. Also the order of list changes between different runs which also makes it harder to find.

If the order of execution causes bugs, there is also merit for debugging in executing tests in a particular order.

PrzemekWirkus, 2016-01-18:
> When you have many test to run, it is difficult to find the test you want in the list

Can you make it more clear? Are you using ```mbedgt --list``` to list tests ?

LiyouZhou, 2016-01-18:
When I want to run a single test, I use mbedgt --list to find the full name of the test. If I run all the test, the report comes back to say which ones failed, I will have to run the failed ones again. Sorting the list just makes the process easier.

PrzemekWirkus, 2016-01-18:
So sorting test results by result?

PrzemekWirkus, 2016-01-18:
I will add missing shuffle test execution order together with this issue fix.

LiyouZhou, 2016-01-18:
I would prefer if test results are sorted by name, so that they are in the same place in every results report.

PrzemekWirkus, 2016-01-19:
OK!

closed 2016-01-22 by PrzemekWirkus

LiyouZhou, 2016-01-22:
@PrzemekWirkus Has this been implemented?