ARMmbed/greentea #79: mbed-greentea doesn't install using python 3

rainierwolfcastle, 2016-02-12:
PS C:\Users\robtay01\Documents\GitHub\testbench-agent> pip install mbed-greentea
Collecting mbed-greentea
  Downloading mbed-greentea-0.1.19.zip (84kB)
    100% |################################| 86kB 3.7MB/s
Collecting PrettyTable>=0.7.2 (from mbed-greentea)
  Downloading prettytable-0.7.2.tar.bz2
Collecting PySerial==2.7 (from mbed-greentea)
  Downloading pyserial-2.7.tar.gz (122kB)
    100% |################################| 122kB 2.3MB/s
Collecting mbed-host-tests>=0.1.18 (from mbed-greentea)
  Downloading mbed-host-tests-0.1.24.zip (102kB)
    100% |################################| 106kB 3.4MB/s
Collecting mbed-ls (from mbed-greentea)
  Downloading mbed-ls-0.1.29.zip (55kB)
    100% |################################| 57kB 4.6MB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "C:\Users\robtay01\AppData\Local\Temp\pip-build-gmyho16q\mbed-ls\setup.py", line 38, in <module>
        long_description=read('README.md'),
      File "C:\Users\robtay01\AppData\Local\Temp\pip-build-gmyho16q\mbed-ls\setup.py", line 33, in read
        return open(os.path.join(os.path.dirname(__file__), fname)).read()
      File "c:\users\robtay01\appdata\local\programs\python\python35\lib\encodings\cp1252.py", line 23, in decode
        return codecs.charmap_decode(input,self.errors,decoding_table)[0]
    UnicodeDecodeError: 'charmap' codec can't decode byte 0x9d in position 11113: character maps to <undefined>

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in C:\Users\robtay01\AppData\Local\Temp\pip-build-gmyho16q\m
bed-ls
You are using pip version 7.1.2, however version 8.0.2 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

PrzemekWirkus, 2016-02-16:
We do not support Python 3.0

closed 2016-02-16 by PrzemekWirkus

0xc0170, 2016-02-16:
@PrzemekWirkus Why did you close it? Wouldn't this be a feature request, to have this module python 3 compatible?