ARMmbed/greentea #90: v0.2.0 fails on mac OS X

bremoran, 2016-03-04:
When I run mbedgt on mac OS X, I get this error:

```
[1457091608.42][HTST][INF] copy image onto target...
[1457091614.67][HTST][INF] starting host test process...
[1457091614.67][HTST][ERR] something went wrong in event main loop!
[1457091614.67][HTST][INF] ==== Traceback start ====
Traceback (most recent call last):
  File "build/bdist.macosx-10.9-x86_64/egg/mbed_host_tests/host_tests_runner/host_test_default.py", line 120, in run_test
    if event_queue.qsize():
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/queues.py", line 143, in qsize
    return self._maxsize - self._sem._semlock._get_value()
NotImplementedError
[1457091614.67][HTST][INF] ==== Traceback end ====
[1457091614.67][HTST][INF] test suite run finished after 0.00 sec...
[1457091614.67][HTST][INF] CONN exited with code: None
Traceback (most recent call last):
  File "/usr/local/bin/mbedhtrun", line 9, in <module>
    load_entry_point('mbed-host-tests==0.2.0', 'console_scripts', 'mbedhtrun')()
  File "build/bdist.macosx-10.9-x86_64/egg/mbed_host_tests/mbedhtrun.py", line 32, in main
  File "build/bdist.macosx-10.9-x86_64/egg/mbed_host_tests/host_tests_runner/host_test_default.py", line 291, in execute
  File "build/bdist.macosx-10.9-x86_64/egg/mbed_host_tests/host_tests_runner/host_test_default.py", line 215, in run_test
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/queues.py", line 143, in qsize
    return self._maxsize - self._sem._semlock._get_value()
NotImplementedError
```

PrzemekWirkus, 2016-03-04:
Similar to: https://github.com/vterron/lemon/issues/11

PrzemekWirkus, 2016-03-07:
See also: https://docs.python.org/2/library/multiprocessing.html#multiprocessing.Queue.qsize:

> Note that this may raise NotImplementedError on Unix platforms like Mac OS X where sem_getvalue() is not implemented.

closed 2016-03-09 by PrzemekWirkus