ARMmbed/greentea #92: Support for option request.txt with host tests

mazimkhan, 2016-03-04:
Host tests are written independent of greentea and htrun. They can use python packages that may not be installed on the CI slaves.
Hence host test developers should be allowed to specify their python package requirements in an optional requirement.txt file. This file can be checked in with the host test script.

PrzemekWirkus, 2016-03-04:
Very good observation!

PrzemekWirkus, 2016-07-25:
@mazimkhan This is still valid requirement but I think it should be satisfied by build system? Please take this under consideration Azim when developing in the future!

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

mazimkhan, 2018-10-02:
Hey @adbridge since host tests are integral to mbed-os this issue can be addressed by putting the requirements in `mbed-os/requirements.txt`. This issue could have been closed when this file was introduced.

closed 2018-10-02 by mazimkhan