ARMmbed/greentea #103: Greentea setup - no sync

0xc0170, 2016-03-10:
The latest changes introduced a new sync command, which works under mbedgt/htrun, but not outside of them. When you just reset the target, or attach a debugger to it, it waits for async command.

The code in the gt client, might help someone if it faces the same problem. This is where an app gets trapped when running outside of htrun/mbedgt:

```
	while (1) {
        greentea_parse_kv(_key, _value, sizeof(_key), sizeof(_value));
        if (strcmp(_key, GREENTEA_TEST_ENV_SYNC) == 0) {
            // Found correct __sunc message
            greentea_send_kv(_key, _value);
            break;
        }
    }
```

Please consider adding debug capabilities - no sync command wait loop. This could be yotta config, so no changing of the code.

Workaround for now - comment out GREENTEA_SETUP in the test

PrzemekWirkus, 2016-04-18:
> Workaround for now - comment out GREENTEA_SETUP in the test

@0xc0170 This is not a good idea because you will not pass things like timeout of host test name to test tools.
Anyway, I will work on this in free time.

PrzemekWirkus, 2016-06-19:
@0xc0170 Martin, I've already added [--sync switch](https://github.com/ARMmbed/htrun/releases/tag/v0.2.18) to `htrun` so we can manipulate how many times `htrun` sends `__sync` packet to device. 
We need to add feature (maybe `-D` + SOMETHING) to enable / disable `__sync` support from DUT and `greente-client`.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

closed 2019-01-07 by bridadan