ARMmbed/greentea #105: AssertionError: __main__ in multiprocessing\forking.py

PrzemekWirkus, 2016-03-10:
There is an occasional ```AssertionError``` on ```c:\python27\lib\multiprocessing\forking.py``` while running Greentea (actual error occurs in ```mbedhtrun```) on Windows.

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "c:\python27\lib\multiprocessing\forking.py", line 380, in main
    prepare(preparation_data)
  File "c:\python27\lib\multiprocessing\forking.py", line 488, in prepare
    assert main_name not in sys.modules, main_name
AssertionError: __main__
```

PrzemekWirkus, 2016-03-10:
Will be fixed in Python 2.7.11: http://bugs.python.org/issue10128

Hokins-Liu, 2016-03-28:
Hi PrezemekWirkus, 
How can I to fix this issue? install a new version of python?
or type upgrade command on yotta?

PrzemekWirkus, 2016-04-18:
This was fixed in ```htrun```:
* https://github.com/ARMmbed/htrun/releases/tag/v0.2.6 and 
* https://github.com/ARMmbed/htrun/pull/74.

closed 2016-04-18 by PrzemekWirkus