ARMmbed/greentea #114: Test spec paths aren't relative to base_path

bridadan, 2016-04-29:
The recently introduce test spec has a field called `base_path`. It can be confusing because it seems like all binary paths would use this as the base path.

For instance, if `base_path: ./build/tests` and my tests are located at `./build/tests/test-case-*.bin`, then I would expect the `path` field for each test case to look like `./test-case-*.bin`. Currently, each test case still has to specify the full path: `./build/tests/test-case-*.bin`.

Furthermore, I would expect the `base_path` would be relative to the location of the test spec file. Instead, the path is relative to the current working directory.

PrzemekWirkus, 2016-05-03:
@mazimkhan Please have a look.

mazimkhan, 2016-05-03:
We have understand the reasons for these fields and take into account the environment where this code is running.
```base_path``` is used to indicate the path where build output is. It is introduced so that tools (greentea) using the test spec can generate artefacts in the same output dir as the binaries.
Yes the the binary ```path``` contains this path and the binary name. It is clear by the name chosen ```path```. 

If the name ```base_path``` is confusing we can rename it rather then putting sense to it by removing relative path from the binary ```path```. The proposal is good as it optimises binary path so that we only have to specify binary name there. But I prefer putting verbose information at the time of generating this spec over processing the path in Greentea. Because this processing will introduce another rule in the spec that image path is ```base_path```/```path```. We should keep it as dumb as possible.

Path is relative from the CWD where Greentea is launched. 
Taking test spec path as root path will introduce assumption that this file is in the root folder. It not wrong as it contains relative paths from root. But it sound circular. Generate test spec in root with paths relative to root -> take root from test spec path. What is wrong in implicit assumption that workspace root i.e. CWD of greentea is root.
Secondly, ```opts.test_spec``` will not be set in the case of Yotta  where we internally generate the test spec object tree.
Please see these comments for #115

bridadan, 2016-05-03:
Ah ok, the main purpose of `base_path` wasn't clear to me. If greentea is creating other artifacts during the test process then it makes sense to have a directory for this.

That being said, it wasn't clear to me that it is assumed that the paths in the test spec need to be relative to the working directory of greentea. The reason was I could specify a full relative or absolute path to the test spec, so I expected that the test spec was ran in the same way, regardless of where greentea was executed.

I see your reasoning now and I think its fine to keep it the way it is, it just needs to be clarified somewhere in the docs and potentially in the CLI. Since this isn't a blocking issue, I'll go ahead and close this issue and related PR.

closed 2016-05-03 by bridadan

mazimkhan, 2016-05-04:
Yes sure I will the update the readme.
One reason I forgot to mention for having relative paths is to support build on one CI slave and execution on another slave.

bridadan, 2016-05-04:
Oh yeah, definitely not saying we need to have absolute paths by default :)

But we do need to be clear if we do have relative paths, what they're relative to.

I suppose we could add a command line option to change the root path of all these relative paths, but we can discuss that later.