ARMmbed/greentea #117: Greentea should wait for mbed device to recover before forking next htrun instance

mazimkhan, 2016-05-05:
Greentea runs htrun is loop for every test. It does not let mbed recover if it crashes.

See logs at http://e108747.cambridge.arm.com:8080/job/gt/3450/console

PrzemekWirkus, 2016-05-05:
Plugins for copy / reset have build in pooling feature.
We can observe next calls to ```mbedhtrun``` are printing information about wrong mount point allocation.
[os.access](https://docs.python.org/2/library/os.html#os.access) function is used to observe mount point for particular device for ~7.5 sec. before error reporting.

```
Plugin info: HostTestPluginCopyMethod_Shell::CopyMethod: Waiting for mount point 'E:' to be ready.................................The system cannot find the drive specified.
```
```
Plugin error: HostTestPluginCopyMethod_Shell::CopyMethod: [ret=1] Command: ['copy', 'build\\frdm-k64f-gcc\\test\\configuration-store-test-misc.bin', 'E:configuration-store-test-misc.bin']
```

See example log:
```
...
02:51:19 mbedgt: selecting test case observer...
02:51:19 	calling mbedhtrun: mbedhtrun -d E: -p COM16:9600 -f ".\build\frdm-k64f-gcc\test\configuration-store-test-misc.bin" -C 4 -c shell -m K64F -t 0240000028884e45001b70116bf000348021000097969900
02:51:19 mbedgt: mbed-host-test-runner: started
02:51:29 [1462413062.67][HTST][INF] copy image onto target...
02:51:44 Plugin info: HostTestPluginCopyMethod_Shell::CopyMethod: Waiting for mount point 'E:' to be ready.................................The system cannot find the drive specified.
02:51:48 Plugin error: HostTestPluginCopyMethod_Shell::CopyMethod: [ret=1] Command: ['copy', 'build\\frdm-k64f-gcc\\test\\configuration-store-test-misc.bin', 'E:configuration-store-test-misc.bin']
...
```

PrzemekWirkus, 2016-05-06:
Response for this issue also for `htrun`:
* Add serial port checks with mbed_lstools before opening port by listener #86 
* Add mount point checks with mbed_lstools before flashing #85 

In most cases `mbedhtrun` is responsible for performing flash-reset-supervise path of testing so above changes + existing guards should be first step of improvements.

PrzemekWirkus, 2016-05-09:
This was delivered with:
* https://github.com/ARMmbed/htrun/pull/85
* https://github.com/ARMmbed/htrun/pull/86
* https://github.com/ARMmbed/htrun/pull/88
and released with [htrun v0.2.8](https://github.com/ARMmbed/htrun/releases/tag/v0.2.8).

closed 2016-05-09 by PrzemekWirkus