ARMmbed/greentea #119: Greentea spawns multiple process for htrun equal to entries in test_spec.json

adbridge, 2016-05-09:
Running mbegt with the following test_spec:
```
{
    "builds": {
        "K64F-ARM": {
            "platform": "K64F",
            "toolchain": "ARM",
            "base_path": "./.build/K64F/ARM",
            "baud_rate": 9600,
            "tests": {
                "case_aynch_validate":{
                    "binaries":[
                        {
                            "binary_type": "bootable",
                            "path": "./.build/K64F/ARM/utest_porting.bin"
                        }
                    ]
                }

            }
        },
        "K64F-GCC": {
            "platform": "K64F",
            "toolchain": "GCC_ARM",
            "base_path": "./.build/K64F/GCC_ARM",
            "baud_rate": 9600,
            "tests": {
                "case_aynch_validate":{
                    "binaries":[
                        {
                            "binary_type": "bootable",
                            "path": "./.build/K64F/GCC_ARM/utest_porting.bin"
                        }
                    ]
                }

            }
        }

    }
}
```
This causes 2 htrun processes to be spawned:
```
        calling mbedhtrun: mbedhtrun -d D: -p COM15:9600:9600 -f "./.build/K64F/ARM/utest_porting.bin" -C 4 -c shell -m K64F -t 02400226a09a1e710000000000000000
000000005d43e3c9
        calling mbedhtrun: mbedhtrun -d D: -p COM15:9600:9600 -f "./.build/K64F/GCC_ARM/utest_porting.bin" -C 4 -c shell -m K64F -t 02400226a09a1e71000000000000
0000000000005d43e3c9
```
both of which try to access the hardware at the same time causing a windows error.

PrzemekWirkus, 2016-05-11:
Good one :D

PrzemekWirkus, 2016-05-11:
Let's use this issue to specify how test_spec::"builds"::[<KEY>::<ALUE>]* behaves.

bridadan, 2016-05-11:
I'd like formally specify the behavior of greentea when it comes to test specs.

# Changes

1. Add a `--test-builds` option, similar in function to the `--test-by-names` command. It accepts a comma separated list of test builds that correspond to test builds in a test spec file. It will run only these specified test builds.
2. Augment the reporter api to support multiple test build runs. For example, for JUnit reports, use the `<testsuite>` key to represent each test build.

# Process

1. Given a test spec file, greentea will examine the file and find all test builds
2. If the `--test-builds` option is specified, filter the test builds to only the list given
3. Execute each valid test build serially
  1. Side note, there could be an option to parallelize test builds as well in the future. This is out of scope for now though.
4. If the `--report-junit` is option is specified, all test build results should be in the one given file. The organization of this file is as follows:
  1. Each `<testsuite>` corresponds to a test build.
      1. Each `<testsuite>` should have the corresponding `platform` and `toolchain` in its `<properties>`. `platform` and `toolchain` correspond one-to-one with the test build's `platform` and `toolchain` properties.
  2. Each `<testcase>` corresponds to a ran test inside of a parent test build (`<testsuite`).

bridadan, 2016-05-17:
@adbridge @mazimkhan @PrzemekWirkus I think this got lost in the hubbub this past week, but any thoughts on this?

adbridge, 2016-05-17:
So I assume the reason for adding --test_builds is to get around the scenario where we instruct the compiler to build ALL the tests and thus they all turn up in the one test_spec.json file? My personal feeling is that having an additional command line flag alongside the test_spec.json seems counter intuitive. I think I might prefer an extra field in the test spec which is something like:
"auto-run-test" : true/false 
When the file is produced all tests default to true, but can downline be manually changed individually.

The alternative solution is that someone takes the default produced file and modifies it to only contain the subset of builds they want to run, they can then rename this to their own name.

I would prefer either of the above to an extra command line list...

adbridge, 2016-05-17:
Of course what might actually be a better solution is a change to the compile chain , where you could specify a list of tests to build (preferably by a .json file or similar) and then you would automatically get the test_spec.json you wanted for your sub test run...

bridadan, 2016-05-17:
@adbridge Yes the reason for the `--test_builds` option was so you could filter from a large test spec a subset of test builds to run.

I would recommend not editing the test spec to control execution. If a user needs to rebuild, then they could blow away all this config when the build system rebuilds the test spec file.

Also, it is my understanding that the test spec file only tells greentea what is available to be flashed, not what it *should* flash. I agree that for complex test campaign, this could be specified by an external file (`--test_campaign=test_campaign.json` perhaps?), but at the moment we aren't there yet. And even in this case, i think we still need to have an equivalent command line argument so we don't force users to create files all the time.

PrzemekWirkus, 2016-05-17:
@adbridge @bridadan I will try to implement this in this PR today.

adbridge, 2016-05-17:
What is wrong with having to create a file? This allows users to produce specific test campaigns and keep them under control. A command line is transient....

PrzemekWirkus, 2016-05-17:
I think there is nothing with creating a file. Brian just had his own opinion :)

PrzemekWirkus, 2016-05-18:
OK, I've updated #120. Added filter for builds from test spec via `-t` which is not used anyways for anything else when using test spec from file (no yotta) so we get to reuse this flag :P 
+ Added some patches after testing stuff.
+ Added few small things to `tests_spec`, @mazimkhan please have a look if everything is OK!

@mazimkhan @bridadan @adbridge Please do code review. Note: some descriptions about fixes are in commit messages. If you have any questions you know what to do! :)

PrzemekWirkus, 2016-05-19:
Fixed with #120.

closed 2016-05-19 by PrzemekWirkus