ARMmbed/greentea #122: Version test spec

mazimkhan, 2016-05-18:
Currently test spec is missing version number. With time we would modify/enhance the test spec. In order maintain compatibility and may be some level of backward compatibility we need to version the test spec.

New test spec version will be instantiated by the build system as that is the source of the spec. Test spec version can not be build system version as spec does not need to change for every new build system version.

Greentea will have hard code to the version(s) that it supports. It can raise error for versions is does not support. Example: "Version p.q.r is no more supported" or "Update Greentea to support version x.y.z".

The less we change test spec version is better. This will keep build system and Greentea compatible most of the times.

PrzemekWirkus, 2016-05-20:
@mazimkhan @adbridge @bridadan:

Few comments from me:
* Test specification is an interface so now we can only (in theory) extend it or add new versions of it.
  * If there is no test spec version in test spec itself we assume THIS (current) version.
  * If there is test spec version specified in the future we will use it as this Issue states.
* Do we know on which level we want to add test specification version?
  * Should it be per build?
  * Should we use semantic versioning?
  * Should we use build system Python semver as version?

```python
{
    "builds": {
        "K64F-ARM": {
            "platform": "K64F",
            "toolchain": "ARM",
            "base_path": "./.build/K64F/ARM",
            ...
            "version": "1.0.0",
            ...
            "tests": {
            ...
            }
        },
        "K64F-GCC": {
            "platform": "K64F",
            "toolchain": "GCC_ARM",
            "base_path": "./.build/K64F/GCC_ARM",
            ...
            "version": "2.1.5",
            ...
            "tests": {
            ...
            }
        }
    }
}
```

bridadan, 2016-05-23:
@PrzemekWirkus Good points! Here is some follow up:

> - Test specification is an interface so now we can only (in theory) extend it or add new versions of it.
>      - If there is no test spec version in test spec itself we assume THIS (current) version.
>      - If there is test spec version specified in the future we will use it as this Issue states.

I agree! +1

> Do we know on which level we want to add test specification version?
> Should it be per build?

I would think this should be a per test spec/file basis, as the top level schema could change as well. So I would move the `version` field to the top level of the test spec object.

> Should we use semantic versioning?

Makes sense to me!

> Should we use build system Python semver as version?

Since any build system could create this file, it seems like we might want to avoid this option. As much as I hate having to create a separate version to keep track of, it seems like the best option at the moment.