ARMmbed/greentea #127: Pretty Dependency Check

BlackstoneEngineering, 2016-05-21:
## Problem
Greentea has many many many dependencies, and if you miss one then the whole thing goes kaboom. 

## Solution
Add UI Enhancement such that each time GT is run it checks for its dependencies, if one is not found then it elegantly exits and pretty prints a message to the user informing them what is missing and suggested remedies to fix it.

PrzemekWirkus, 2016-05-23:
Can you give example of dependency which is not included in `setup.py` ? I wanna fix it asap so people installing our tools do not "kaboom"!

BlackstoneEngineering, 2016-05-23:
Dont remember what spawned this, think it was colorama on @sg- 's machine the other day, though that seems to have been fixed in the dependency (just checked latest setup.py dependencies and it looks to be there). This is more a futureproofing safeguard now that greentea is coming more to the foreground.

PrzemekWirkus, 2016-05-24:
colorama is there; https://github.com/ARMmbed/greentea/blob/master/setup.py#L61
Probably something else `setup.py` do not specify. I will have a look on imports inside greentea, htrun and mbed-ls.

BlackstoneEngineering, 2016-05-24:
Kk, for reference it was on a stock mac. So maybe try borrowing @iriark01 's machine and try running it to see if it works?

-Austin

iriark01, 2016-05-24:
I had this issue when I first documented Greentea - I had to manually install a lot of stuff to get it up and running. At the time I added them to the prerequisites list in the guide - maybe that's still available somewhere.

PrzemekWirkus, 2016-06-16:
@iriark01 Do you remember stuff you had to install? Is it still reproducable?
Are you maybe mistaking yotta dependencies like CMake with that [Greentea requires](https://github.com/ARMmbed/greentea/blob/master/setup.py#L54).

iriark01, 2016-06-16:
It was this (can't rule out the possibility that some of it was yotta):

Please install the following:

__General tools__

* [Python](https://www.python.org/downloads/). If you do not have Python installed already, we recommend [version 2.7.9](https://www.python.org/downloads/release/python-279/). You'll need to add the following modules:

	* [Pip](https://pypi.python.org/pypi/pip). Pip comes bundled with some Python versions; run ``$ pip --version`` to see if you already have it.

	* [setuptools](https://pythonhosted.org/an_example_pypi_project/setuptools.html) to install dependencies.

	* cryptography (install using ``pip``).

	* hgapi (install using ``pip``).

	* colorama (install using ``pip``).

	* PyGithub (install using ``pip``).

	* semantic_version==2.4.1 (install using ``pip``) - note that it requires [Microsoft Visual C++ 9.0](http://www.microsoft.com/en-gb/download/details.aspx?id=44266).

	* project-generator==0.5.7 (install using ``pip``).

	* pyOCD (install using ``pip``).

* The ``cp`` shell command must be available to flash certain boards. It is sometimes available by default, for example on Linux, or you can install the [Git command line tools](https://github.com/github/hub).

* [Grep](http://gnuwin32.sourceforge.net/packages/grep.htm).

* [yotta](https://github.com/ARMmbed/yotta): used to build tests from the mbed SDK. Please note that **yotta has its own set of dependencies**, listed in the [installation instructions](http://armmbed.github.io/yotta/#installing-on-windows).

__Windows only__

* If your OS is Windows, please follow the installation instructions [for the serial port driver](https://developer.mbed.org/handbook/Windows-serial-configuration).

__Required modules__

* The mbed SDK sources. These are provided in the release sources under the **libraries/mbed-drivers** directory.

* mbed-ls: installation instructions can be found [in the repository](https://github.com/ARMmbed/mbed-ls#installation-from-python-sources).

* htrun: installation instructions can be found [in the respository](https://github.com/ARMmbed/htrun#installation-from-python-sources).

PrzemekWirkus, 2016-06-16:
Yes, most of this tools are not required by Greentea, like yotta, grep etc. Other like Python related are described in installation chapter.

I think I should concentrate on dependecies Greentea has on other tools.
I will discard yotta as I do not think yotta should be installed with Greeentea together. Even it is a command line tool Greentea may call if no `--skip-build` is specified.

closed 2016-06-16 by PrzemekWirkus