ARMmbed/greentea #131: Remove dependency on mbed_target_info.py

bridadan, 2016-05-26:
Currently if any platform that is tested does not exist in `mbed_target_info.py`, then you get an error: `Platform <platform> not in target info store.`

This is only used when testing yotta modules, but it is not necessary when using a test spec file. In this case, greentea should make its best effort to test all platforms.

It looks the only info it's using from the target info store in the case of a test spec file is `program_cycle_s`. Ideally we would use a different mechanism for checking to see if the platform is ready. Though for the moment I think using a sensible default would be fine.

PrzemekWirkus, 2016-05-27:
I will handle this one.

PrzemekWirkus, 2016-05-27:
@bridadan Please have a look at #133 and review / test if it works for you.

bridadan, 2016-06-02:
Looks good!

closed 2016-06-02 by bridadan