ARMmbed/greentea #150: ctrl+C when running tests doesn't stop the tests (Linux)

mlnx, 2016-06-23:
- Type: Bug
- Priority:  Major

Raised earlier in wrong repo: https://github.com/ARMmbed/mbed-os/issues/387

## Bug

Under Linux Ubuntu 14.04, the command ctrl+C doesn't (fully) stop the tests. To reproduce the problem, just run `mbed test` under linux and try to stop it by using ctrl+C.

I get access to the console, but tests continue to run.

The processes continue to run in the background:
```
19651 pts/12   00:00:01 mbedgt
21283 pts/12   00:00:02 mbedhtrun
21299 pts/12   00:00:00 mbedhtrun
```
The workaround is to manually kill the processes.

PrzemekWirkus, 2016-06-23:
@mlnx Please let me know following:
```
$ mbed-ls --version
$ mbed-host-tests --version
$ mbed-greentea --version
$ python --version
```

mlnx, 2016-06-23:
```
$ mbedls --version
0.2.13
$ mbedhtrun --version
0.2.18
$ mbedgt --version
0.2.24
$ python --version
Python 2.7.6
```

bridadan, 2017-06-06:
This should be fixed in #226, feel free to try out the fix if you like.

theotherjimmy, 2017-09-20:
Closing as it's fixed.

closed 2017-09-20 by theotherjimmy