ARMmbed/greentea #157: [minor] Reported passes vs failures miscount info

mlnx, 2016-06-29:
Please treat this as low priority. I think mbedgt is showing info which may not be very clear to the user.

Unless I see the code, it's not clear what's the relation among the numbers: (a, b) vs (c,d)
https://github.com/ARMmbed/greentea/blob/9efdb07420715c7ec7f4071dc465cd01c99eb687/mbed_greentea/mbed_greentea_cli.py#L555

Maybe this could be simplified in a single line?

```
mbedgt: test case summary mismatch: reported passes vs failures miscount!
        (47, 0) vs (41, 0)    <---
```

PrzemekWirkus, 2016-06-30:
This messages relates to utest reporting different number of passes, failures in log and in summary.
I will change that so it is more verbose for the user. In the mean time please paste whole log from this test run. Thanks!

closed 2016-07-03 by PrzemekWirkus

mlnx, 2016-07-06:
Find below the relevant part of the log.
I believe the PR (https://github.com/ARMmbed/greentea/pull/159) clarifies things for the user now.
Thanks!

```
mbedgt: test case summary: 30 passes, 0 failures
mbedgt: test case summary mismatch: reported passes vs failures miscount!     <-------- RED
        (30, 0) vs (17, 0)     <--------
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found (to directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000028634e45003c5002673a0007a101000097969900
mbedgt: test suite 'mbed-os-features-storage-tests-cfstore-create' ................................... OK in 18.21 sec
        test case: 'CREATE_test_00' .................................................................. OK in 0.09 sec
        test case: 'CREATE_test_01_end' .............................................................. OK in 0.04 sec
        test case: 'CREATE_test_01_start' ............................................................ OK in 0.05 sec
        test case: 'CREATE_test_02_end' .............................................................. OK in 0.06 sec
        test case: 'CREATE_test_02_start' ............................................................ OK in 0.05 sec
        test case: 'CREATE_test_03_end' .............................................................. OK in 1.46 sec
        test case: 'CREATE_test_03_start' ............................................................ OK in 0.04 sec
        test case: 'CREATE_test_04_end' .............................................................. OK in 0.11 sec
        test case: 'CREATE_test_04_start' ............................................................ OK in 0.05 sec
        test case: 'CREATE_test_05_end' .............................................................. OK in 0.11 sec
        test case: 'CREATE_test_05_start' ............................................................ OK in 0.05 sec
        test case: 'CREATE_test_06_end' .............................................................. OK in 0.05 sec
        test case: 'CREATE_test_06_start' ............................................................ OK in 0.05 sec
mbedgt: test case summary: 13 passes, 0 failures
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found (to directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000028634e45003c5002673a0007a101000097969900
mbedgt: test suite 'mbed-os-tests-mbed_drivers-wait_us' .............................................. OK in 22.10 sec
        test case: 'Timers: wait_us' ................................................................. OK in 10.13 sec
mbedgt: test case summary: 1 pass, 0 failures
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found (to directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000028634e45003c5002673a0007a101000097969900
mbedgt: test suite 'mbed-os-features-storage-tests-cfstore-find2' .................................... OK in 14.19 sec
        test case: 'FIND2_test_00' ................................................................... OK in 0.08 sec
        test case: 'FIND2_test_01' ................................................................... OK in 0.04 sec
        test case: 'FIND2_test_02' ................................................................... OK in 0.04 sec
mbedgt: test case summary: 3 passes, 0 failures
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found (to directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: mbed-os-tests-mbed_drivers-dev_null
        test case: mbed-os-tests-mbed_drivers-dev_null
mbedgt: test on hardware with target id: 0240000028634e45003c5002673a0007a101000097969900
mbedgt: test suite 'mbed-os-tests-mbed_drivers-dev_null' ............................................. OK in 14.78 sec
        test case: 'mbed-os-tests-mbed_drivers-dev_null' ............................................. OK in 14.78 sec
mbedgt: test case summary: 1 pass, 0 failures
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found (to directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000028634e45003c5002673a0007a101000097969900
mbedgt: test suite 'mbed-os-features-storage-tests-flash_journal-basicapi' ........................... OK in 58.44 sec
        test case: 'commit without logs' ............................................................. OK in 0.10 sec
        test case: 'failed small write followed by padded write' ..................................... OK in 0.15 sec
        test case: 'initialize' ...................................................................... OK in 0.04 sec
[... cut ...]
        test case: 'reset and initialize6' ........................................................... OK in 0.22 sec
mbedgt: test case summary: 47 passes, 0 failures
mbedgt: test case summary mismatch: reported passes vs failures miscount!    <-------- RED
        (47, 0) vs (41, 0)   <--------
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found (to directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
```