ARMmbed/greentea #162: Provide a mechanism to change baud rate

bremoran, 2016-07-07:
In some tests, it is desirable to change baud rates. It would be nice if mbedgt (or one of its subtools) and greentea-client were to provide a mechanism for this. When a baud rate change is initiated, it is likely necessary for mbedgt to close the serial port, reopen with the new parameters, then initiate a new sync.

cc @PrzemekWirkus

PrzemekWirkus, 2016-07-07:
This feature will require us to change a bit `htrun` and `greentea-client`. We will have to provide inetrface (events) to change baudrate and handshake after reconnection.
Main changes may include:
* Add new event to baudrate exchange from DUT, in `greentea-client`
* Add handle for event "request to change baudrate", in `htrun`
  * Note: this event would be available for all host tests inherited from base host test class
  * Generic API should be provided on `htrun` / host test side
* No dependency on `mbed-os` or other, so each OS should use its own way to handshake after reconnection.

bridadan, 2017-06-06:
This is a duplicate of #196. @adbridge @mazimkhan Can we please close this one?

closed 2017-06-07 by adbridge