ARMmbed/greentea #167: Move / remove extraneous messages from top level of tests

BlackstoneEngineering, 2016-07-13:
## Problem
Running `mbed tests` gives too much output. 

For example: 
```
$ mbed test -n tests-api-uart

Building library mbed-build (K64F, GCC_ARM)
Building project GCC_ARM to tests-api-spi (K64F, GCC_ARM)
+----------------------------+-------+-------+------+
| Module                     | .text | .data | .bss |
+----------------------------+-------+-------+------+
| Fill                       |    85 |     3 | 2536 |
| Misc                       | 37480 |  2224 | 2312 |
| core/hal                   | 16051 |    16 |  565 |
| core/rtos                  |  6815 |    24 | 2662 |
| features/FEATURE_IPV4      |    96 |     0 |   48 |
| frameworks/greentea-client |  1580 |    56 |   52 |
| frameworks/unity           |   584 |     0 |  160 |
| frameworks/utest           |  5312 |     5 |  805 |
| Subtotals                  | 68003 |  2328 | 9140 |
+----------------------------+-------+-------+------+
Allocated Heap: 65540 bytes
Allocated Stack: 32768 bytes
Total Static RAM memory (data + bss): 11468 bytes
Total RAM memory (data + bss + heap + stack): 109776 bytes
Total Flash memory (text + data + misc): 71371 bytes
Image: .build/tests/K64F/GCC_ARM/TESTS/API/SPI/tests-api-spi.bin

Build successes:
  * K64F::GCC_ARM::MBED-BUILD
  * K64F::GCC_ARM::TESTS-API-SPI
mbedgt: test specification file 'c:\cygwin\home\ausbla01\mbed\ci-test-shield\.build/tests\K64F\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'c:\cygwin\home\ausbla01\mbed\ci-test-shield\.build/tests\K64F\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	detected 'K64F' -> 'K64F[0]', console at 'COM165', mounted at 'D:', target id '0240000025354e450025400d47d00046bb91000097969900'
mbedgt: processing target 'K64F' toolchain 'GCC_ARM' compatible platforms...
mbedgt: using platform 'K64F' for test:
	target_id_mbed_htm = '0240000025354e450025400d47d00046bb91000097969900'
	daplink_hic_id = '97969900'
	mount_point = 'D:'
	daplink_automation_allowed = '0'
	daplink_daplink_mode = 'Interface'
	daplink_local_mods = '0'
	daplink_unique_id = '0240000025354e450025400d47d00046bb91000097969900'
	daplink_interface_crc = '0x33e2dd18'
	target_id = '0240000025354e450025400d47d00046bb91000097969900'
	serial_port = 'COM165:9600'
	target_id_usb_id = '0240000025354e450025400d47d00046bb91000097969900'
	platform_name = 'K64F'
	platform_name_unique = 'K64F[0]'
	daplink_git_sha = '34182e2cce4ca99073443ef29fbcfaab9e18caec'
	daplink_interface_version = '0241'
	daplink_auto_reset = '0'
	daplink_usb_interfaces = 'MSD, CDC, HID'
	daplink_version = '0241'
mbedgt: test case filter (specified with -n option)
	test filtered in 'tests-api-spi'
mbedgt: running 1 test for platform 'K64F' and toolchain 'GCC_ARM'
	use 1 instance for testing
mbedgt: checking for 'host_tests' directory above image directory structure
	'host_tests' directory not found (to directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000025354e450025400d47d00046bb91000097969900
mbedgt: test suite 'tests-api-spi' ................................................................... FAIL in 12.09 sec
	test case: 'Testing SPI via SD Write' ........................................................ FAIL in 0.14 sec
mbedgt: test case summary: 0 passes, 1 failure
mbedgt: all tests finished!
[... 118 lines cut ...]
| frameworks/greentea-client |  1580 |    56 |   52 |
| frameworks/unity           |   584 |     0 |  160 |
| frameworks/utest           |  5312 |     5 |  805 |
| Subtotals                  | 68003 |  2328 | 9140 |
+----------------------------+-------+-------+------+
Allocated Heap: 65540 bytes
Allocated Stack: 32768 bytes
Total Static RAM memory (data + bss): 11468 bytes
Total RAM memory (data + bss + heap + stack): 109776 bytes
Total Flash memory (text + data + misc): 71371 bytes
Build Success!

mbedgt: detecting connected mbed-enabled devices...
mbedgt: Detected platform 'K64F' mounteed at 'D:' on 'COM165:9600'
mbedgt: Running test suite 'tests-api-spi' ... . . . . . . . . . .
mbedgt: test case report:
+--------------+---------------+---------------+--------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite    | test case                | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+---------------+--------------------------+--------+--------+--------+--------------------+
| K64F-GCC_ARM | K64F          | tests-api-spi | Testing SPI via SD Write | 0      | 1      | FAIL   | 0.14               |
+--------------+---------------+---------------+--------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 FAIL
mbedgt: completed in 12.16 sec

```

BlackstoneEngineering, 2016-07-13:
Also colorize the Failures to red, colorize the passes to Green, bold the important bits around number of failures / sucecesses, total flash size, and propogate any errors from within to the top level as sited here :  https://github.com/ARMmbed/greentea/issues/166

BlackstoneEngineering, 2016-07-13:
@bridadan  over here

PrzemekWirkus, 2016-07-14:
@BlackstoneEngineering `pip install mbed-greentea -U` -> Interface looks a bit different now (extensive device detection data is gone, and replaced with table).

PrzemekWirkus, 2016-07-14:
`mbed test` command calls Greentea with extra flags (I assume it uses Greentea's `-V` for extra verboseness).

Without `-V` flag one test case full run looks like this:
```
mbedgt -n tests-mbed_drivers-generic_tests
mbedgt: greentea test automation tool ver. 1.1.1
mbedgt: using multiple test specifications from current directory!
        using '.build\tests\K64F\GCC_ARM\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                                        |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | K64F          | K64F[0]              | COM229      | G:          | 0240000033514e450040500585d4000be981000097969900 |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
mbedgt: processing target 'K64F' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                                        |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
        | K64F          | K64F[0]              | COM229:9600 | G:          | 0240000033514e450040500585d4000be981000097969900 |
        +---------------+----------------------+-------------+-------------+--------------------------------------------------+
mbedgt: test case filter (specified with -n option)
        test filtered in 'tests-mbed_drivers-generic_tests'
mbedgt: running 1 test for platform 'K64F' and toolchain 'GCC_ARM'
        use 1 instance for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000033514e450040500585d4000be981000097969900
mbedgt: test suite 'tests-mbed_drivers-generic_tests' ................................................ OK in 13.15 sec
        test case: 'Basic' ........................................................................... OK in 0.03 sec
        test case: 'Blinky' .......................................................................... OK in 0.03 sec
        test case: 'C++ heap' ........................................................................ OK in 0.09 sec
        test case: 'C++ stack' ....................................................................... OK in 0.15 sec
mbedgt: test case summary: 4 passes, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.2474493506
mbedgt: test suite report:
+--------------+---------------+----------------------------------+--------+--------------------+-------------+
| target       | platform_name | test suite                       | result | elapsed_time (sec) | copy_method |
+--------------+---------------+----------------------------------+--------+--------------------+-------------+
| K64F-GCC_ARM | K64F          | tests-mbed_drivers-generic_tests | OK     | 13.15              | shell       |
+--------------+---------------+----------------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+--------------+---------------+----------------------------------+-----------+--------+--------+--------+--------------------+
| target       | platform_name | test suite                       | test case | passed | failed | result | elapsed_time (sec) |
+--------------+---------------+----------------------------------+-----------+--------+--------+--------+--------------------+
| K64F-GCC_ARM | K64F          | tests-mbed_drivers-generic_tests | Basic     | 1      | 0      | OK     | 0.03               |
| K64F-GCC_ARM | K64F          | tests-mbed_drivers-generic_tests | Blinky    | 1      | 0      | OK     | 0.03               |
| K64F-GCC_ARM | K64F          | tests-mbed_drivers-generic_tests | C++ heap  | 1      | 0      | OK     | 0.09               |
| K64F-GCC_ARM | K64F          | tests-mbed_drivers-generic_tests | C++ stack | 1      | 0      | OK     | 0.15               |
+--------------+---------------+----------------------------------+-----------+--------+--------+--------+--------------------+
mbedgt: test case results: 4 OK
mbedgt: completed in 13.31 sec
```

PrzemekWirkus, 2016-07-14:
I can suppress prints:
```
mbedgt: checking for 'host_tests' directory above image directory structure
        'host_tests' directory not found: two directory levels above image path checked
```
outside `-V` mode.

PrzemekWirkus, 2016-07-14:
@BlackstoneEngineering 

> Also colorize the Failures to red, colorize the passes to Green, bold the important bits around number of failures / sucecesses

How about command line switch `--christmas-tree-cloros` ? ;)

bridadan, 2016-07-14:
> How about command line switch --christmas-tree-colours ? ;)
+1 for different color themes XD (I kid!)
But colorizing failures/passes would be pretty awesome!

`mbed test` by default will not add the `-V` tag to greentea. It will be added if `-v` is supplied to `mbed test`

PrzemekWirkus, 2016-07-25:
Colorizing is a bigger pile of work because colors are available only for logger, not for core of the Greentea.
So I will close this Issue with regards to changes in #169 - let's have one issue, one problem. This Issue was more about extensive logging rather than colorizing. See Issue subject **Move / remove extraneous messages from top level of tests**.

closed 2016-07-25 by PrzemekWirkus