ARMmbed/greentea #168: IOERR_SERIAL

cyliangtw, 2016-07-14:
To execute " mbed test -m K64F -t ARM -n mbed-os-tests-integration-basic" and got IOERR_SERIAL.
In actually, I wrote one simple test code to ensure open & read serial port well by "python.exe ../serial_test.py".
The details of mbed test log as below:
+----------+---------------+---------------------------------+--------------+--------------------+----+
| target   | platform_name | test suite                      | result       | elapsed_time (sec) | copy_method |
+----------+---------------+---------------------------------+--------------+--------------------+--+
| K64F-ARM | K64F          | mbed-os-tests-integration-basic | IOERR_SERIAL | 7.91               | shell       |
+----------+---------------+---------------------------------+--------------+--------------------+--+
mbedgt: test suite results: 1 IOERR_SERIAL
mbedgt: test case report:

PrzemekWirkus, 2016-07-14:
@cyliangtw Can you give me more details what mbed-OS you are actually executing?
* Which OS you are using?
* Which version (SHA) of OS you are using?
* Which tools versions you are using:
```
$ mbedgt --version
$ mbedhtrun --version
$ mbedls --version
$ mbed --version
```

`IOERR_SERIAL` error can appear when you for example have:
* old version of DAPlink software flashed on your K64F (very likely),
* no access to serial port (port is already opened), 
* you do not have privileges to open serial port or just transmission is corrupted.

Also can you plug your `K64F` and execute command:
```
$ mbedls
```
I'm interested in its output from that command and in particular column `daplink_version`.

You can call Greentea test suite yourself from within mbed-os directory, use flag `-V` to add extra verboseness and copy-paste results here:
```
~/mbed-os$ mbed test --compile -m K64F -t ARM
```
```
~/mbed-os$ mbedgt -V -t K64F-ARM -n tests-integration-basic
```
```
mbedgt: greentea test automation tool ver. 1.1.1
mbedgt: using multiple test specifications from current directory!
	using '.build\tests\K64F\ARM\test_spec.json'
	using '.build\tests\K64F\GCC_ARM\test_spec.json'
	using '.build\tests\K64F\IAR\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
	+---------------+----------------------+-------------+-------------+--------------------------------------------------+
	| platform_name | platform_name_unique | serial_port | mount_point | target_id                                        |
	+---------------+----------------------+-------------+-------------+--------------------------------------------------+
	| K64F          | K64F[0]              | COM229      | G:          | 0240000033514e450040500585d4000be981000097969900 |
	+---------------+----------------------+-------------+-------------+--------------------------------------------------+
mbedgt: processing target 'K64F' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
	+---------------+----------------------+-------------+-------------+--------------------------------------------------+
	| platform_name | platform_name_unique | serial_port | mount_point | target_id                                        |
	+---------------+----------------------+-------------+-------------+--------------------------------------------------+
	| K64F          | K64F[0]              | COM229:9600 | G:          | 0240000033514e450040500585d4000be981000097969900 |
	+---------------+----------------------+-------------+-------------+--------------------------------------------------+
mbedgt: test case filter (specified with -n option)
	test filtered in 'tests-integration-basic'
mbedgt: running 1 test for platform 'K64F' and toolchain 'ARM'
	use 1 instance for testing
mbedgt: checking for 'host_tests' directory above image directory structure
	'host_tests' directory not found: two directory levels above image path checked
mbedgt: selecting test case observer...
	calling mbedhtrun: mbedhtrun -m K64F -p COM229:9600 -f ".build/tests/K64F/ARM/TESTS/integration/basic/tests-integration-basic.bin" -d G: -C 4 -c shell -t 0240000033514e450040500585d4000be981000097969900
mbedgt: mbed-host-test-runner: started
[1468498293.06][HTST][INF] host test executor ver. 1.1.0
[1468498293.06][HTST][INF] copy image onto target...
[1468498293.06][COPY][INF] Waiting up to 60 sec for '0240000033514e450040500585d4000be981000097969900' mount point (current is 'G:')...
        1 file(s) copied.
[1468498300.36][HTST][INF] starting host test process...
[1468498300.90][CONN][INF] starting connection process...
[1468498300.90][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1468498300.90][CONN][INF] initializing serial port listener...
[1468498300.90][HTST][INF] setting timeout to: 60 sec
[1468498300.90][SERI][INF] serial(port=COM229, baudrate=9600, timeout=0)
[1468498300.90][PLGN][INF] Waiting up to 60 sec for '0240000033514e450040500585d4000be981000097969900' serial port (current is 'COM229')...
[1468498300.94][SERI][INF] reset device using 'default' plugin...
[1468498301.20][SERI][INF] waiting 1.00 sec after reset
[... 37 lines cut ...]
	no test case report present, assuming test suite to be a single test case!
	test suite: tests-integration-basic
	test case: tests-integration-basic
mbedgt: test on hardware with target id: 0240000033514e450040500585d4000be981000097969900
mbedgt: test suite 'tests-integration-basic' ......................................................... OK in 9.98 sec
	test case: 'tests-integration-basic' ......................................................... OK in 9.98 sec
mbedgt: test case summary: 1 pass, 0 failures
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.2807166858
mbedgt: test suite report:
+----------+---------------+-------------------------+--------+--------------------+-------------+
| target   | platform_name | test suite              | result | elapsed_time (sec) | copy_method |
+----------+---------------+-------------------------+--------+--------------------+-------------+
| K64F-ARM | K64F          | tests-integration-basic | OK     | 9.98               | shell       |
+----------+---------------+-------------------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 OK
mbedgt: test case report:
+----------+---------------+-------------------------+-------------------------+--------+--------+--------+--------------------+
[... cut ...] assed | failed | result | elapsed_time (sec) |
+----------+---------------+-------------------------+-------------------------+--------+--------+--------+--------------------+
| K64F-ARM | K64F          | tests-integration-basic | tests-integration-basic | 1      | 0      | OK     | 9.98               |
+----------+---------------+-------------------------+-------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 OK
mbedgt: completed in 10.29 sec
```

cyliangtw, 2016-07-14:
OS: Win7
Tools versions :
```
$ mbedgt 1.1.1
$ mbedhtrun 1.1.0
$ mbedls 1.1.1
$ mbed 0.8.2
```
>•	no access to serial port (port is already opened), 
>•	you do not have privileges to open serial port or just transmission is corrupted.
Ans: I verified open & read serial port by "python.exe ../serial_test.py" successfully, so not suffered the above 2 cases.

`[cygwin~mbed-os]$ mbedls`
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| K64F          | K64F[0]              | D:          | COM1        | 02400201C30D4E723EF3B3CA | 0201            |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+

```
[cygwin~mbed-os]$ mbedgt -V -t K64F-ARM -n tests-integration-basic
mbedgt: greentea test automation tool ver. 1.1.1
mbedgt: greentea should be run inside a Yotta module or --test-spec switch should be used.
mbedgt: completed in 0.00 sec
mbedgt: exited with code -1
[cygwin~mbed-os]$

```

PrzemekWirkus, 2016-07-14:
@cyliangtw Your K64F daplink version is super-old ;) You have `0201`, current stable is [0241](https://github.com/mbedmicro/DAPLink/releases/tag/v0241).

PrzemekWirkus, 2016-07-14:
@cyliangtw You can find instructions how to flash K64F with new firmware [here](https://developer.mbed.org/handbook/Firmware-FRDM-K64F).

Latest binary to flash your K64F you can find inside [DAPLink/0241_release_package_34182e2c.zip](https://github.com/mbedmicro/DAPLink/releases/download/v0241/0241_release_package_34182e2c.zip). Filename is *0241_k20dx_frdmk64f_0x5000.bin*.

PrzemekWirkus, 2016-07-14:
I'm also surprised with this log:
```
mbedgt: greentea should be run inside a Yotta module or --test-spec switch should be used.
```

Can you: `[cygwin~mbed-os]$ ls` for me?

cyliangtw, 2016-07-14:
It's early access version mbed-os-5.0.1-epr .
```
[cygwin~mbed-os]$
[cygwin~mbed-os]$ ls
circle.yml  CONTRIBUTING.md  core  core.lib  docs  features  frameworks  Jenkinsfile  LICENSE  README.md  TESTS  travis
[cygwin~mbed-os]$

```

cyliangtw, 2016-07-14:
After applied  0241_k20dx_frdmk64f_0x5000.bin, IOERR_SERIAL issue is resolved.
Thanks a lot.

closed 2016-07-14 by cyliangtw