ARMmbed/greentea #176: Doc is still about yotta tools

RdaZhongyao, 2016-08-06:
After mbed OS 5.1 released, test tools should be merged to it, I think.
However, this repo's doc is still about yotta, not mbed-cli.

bridadan, 2016-08-08:
@RdaZhongyao Thanks for bringing up this issue. Greentea is still compatible with yotta (in fact, it can be used with any build system now with the introduction of the [test specification file](https://github.com/ARMmbed/greentea#test-specification-json-formatted-input)), but I agree that the docs could use some updates in regards to the mbed OS 5.1 release.

cc @mazimkhan

ivan-adzic-tridonic-com, 2016-08-19:
@daniel-cesarini-tridonic-com @markus-becker-tridonic-com this is relevant also for us.

AnotherButler, 2017-04-13:
I believe PR #217 resolved this issue. @mazimkhan @adbridge Can you please confirm?

mazimkhan, 2017-04-18:
Yes #217 fixes this.

bridadan, 2017-04-18:
Should be good to close this then @mazimkhan

closed 2017-04-18 by mazimkhan