ARMmbed/greentea #179: Build and Test with mbedgt

cuevasj, 2016-08-10:
I was under the impression that the command mbedgt would both build (by calling yotta build) and test the application.  However when I run mbedgt -V without first calling yotta build, it says “no platform/target matching tests were found!” (I assume because they weren’t built) and exits with code -10.  If I run yotta build then mbedgt -V, the tests are run as expected.  Text from the command prompt ist attached.
[log_mbedgt-V.txt](https://github.com/ARMmbed/greentea/files/411356/log_mbedgt-V.txt)

Am I understanding the command correctly?  If not how should I use it to build and test in one command?

Notes
I’m working with the bed-drivers repository.
I did set the yotta target to frdm-k64f-gcc before calling mbedgt.

cuevasj, 2016-08-10:
The version I'm using are:
mbed-greentea==1.2.0
mbed-host-tests==1.1.0
mbed-ls==1.2.0
mbed-test-wrapper==1.0.0
yotta 0.16.4

bridadan, 2016-08-10:
@cuevasj Building the tests by default was the behavior in mbed OS 3, and it was my impression that we were looking to maintain this compatibility in greentea if at all possible. @mazimkhan Do you know if removing the build step for yotta was intentional or did it just slip through? If this was intentional, we'll need to update the docs to reflect that. And we could also clean up a lot of the arguments in the CLI.

With the most recent release, we were trying to decouple greentea from build systems by adding the [test spec file](https://github.com/ARMmbed/greentea#test-specification-json-formatted-input), and I imagine we'll probably continue down this road.

In the meantime, the workaround you posted is definitely valid. Sorry about the inconvenience!

mazimkhan, 2016-08-11:
@cuevasj Brian is right. We were ending up in a situation where greentea can build and yotta can test. This kind of versatility looks cool but also adds dependencies. 
When mbed OS 5.1 work started we had to enable greentea on non Yotta mbed OS project. So we moved away from building from greentea. Tools are better when they do one thing and one thing  right.

cuevasj, 2016-08-24:
So I should close this issue?

bridadan, 2016-08-25:
@cuevasj Yes this issue is ok to close

bridadan, 2017-06-06:
Good to close @mazimkhan @adbridge

closed 2017-06-07 by adbridge