ARMmbed/greentea #180: Problem in processing mbedgt with custom targets on mbedos5

ivan-adzic-tridonic-com, 2016-08-19:
We are trying to run testcases on NUCLEO_F091RC board, where we have created custom target TRI_SE (in targets.json of mbed-os/hal/), which inherits NUCLEO_F091RC, toolchain is GCC_ARM.

We build tests with **TRI_SE** ::
```
$ mbed test -n tests* -m TRI_SE -t GCC_ARM
```
it has failed, partial output is:

```
Build successes:
  * TRI_SE::GCC_ARM::MBED-BUILD
  * TRI_SE::GCC_ARM::TESTS-STC-580-TESTCASE-1
  * TRI_SE::GCC_ARM::TESTS-STC-580-TESTCASE-2
  * TRI_SE::GCC_ARM::TESTS-STC-580-TESTCASE-3
mbedgt: greentea test automation tool ver. 1.2.0
mbedgt: test specification file 'C:\Projects\mbedos5\gl-apps\n4m-tests\.build/tests\TRI_SE\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Projects\mbedos5\gl-apps\n4m-tests\.build/tests\TRI_SE\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'TRI_SE' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.4007050724
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.14 sec
mbedgt: exited with code -110
[mbed] ERROR: "mbedgt" returned error code -110.
[mbed] ERROR: Command "mbedgt --test-spec C:\Projects\mbedos5\gl-apps\n4m-tests\.build/tests\TRI_SE\GCC_ARM\test_spec.json -n tests*" in "C:\Projects\mbedos5\gl-apps\n4m-tests"
---

```

Hint:
When we build same tests with **NUCLEO_F091RC** :
```
$ mbed test -n tests* -m NUCLEO_F091RC -t GCC_ARM
```
it successes, and we have got:
```
Build successes:
  * NUCLEO_F091RC::GCC_ARM::MBED-BUILD
  * NUCLEO_F091RC::GCC_ARM::TESTS-STC-580-TESTCASE-1
  * NUCLEO_F091RC::GCC_ARM::TESTS-STC-580-TESTCASE-2
  * NUCLEO_F091RC::GCC_ARM::TESTS-STC-580-TESTCASE-3
mbedgt: greentea test automation tool ver. 1.2.0
mbedgt: test specification file 'C:\Projects\mbedos5\gl-apps\n4m-tests\.build/tests\NUCLEO_F091RC\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Projects\mbedos5\gl-apps\n4m-tests\.build/tests\NUCLEO_F091RC\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'NUCLEO_F091RC' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        test filtered in 'tests-stc-580-testcase-1'
        test filtered in 'tests-stc-580-testcase-2'
        test filtered in 'tests-stc-580-testcase-3'
mbedgt: running 3 tests for platform 'NUCLEO_F091RC' and toolchain 'GCC_ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test case summary event not found
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-stc-580-testcase-1
        test case: tests-stc-580-testcase-1
mbedgt: test on hardware with target id: 07500200667370285916E645
mbedgt: test suite 'tests-stc-580-testcase-1' ........................................................ OK in 11.68 sec
        test case: 'tests-stc-580-testcase-1' ........................................................ OK in 11.68 sec
mbedgt: test case summary: 1 pass, 0 failures

```

Could you please guide us how to process custom targets with mbedos5 and greentea tests.
@MarceloSalazar 

cc @daniel-cesarini-tridonic-com and @markus-becker-tridonic-com

mazimkhan, 2016-08-19:
Hi,

Greentea uses mbed-ls to detect platforms. mbed-ls has a feature to mock new/custom targets. Use following command in the path you are running greentea.
```
mbedls -m 1234:TRI_SE 
```

Above replace ```1234``` with first four chars of your boards target Id. This will enable mbedls and hence Greentea to detect your target and execute the tests.

ivan-adzic-tridonic-com, 2016-08-19:
Hi,
we already have done the following:
```
$ mbedls -m 0750:TRI_SE
```
the result is:
```
$ mbedls
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| TRI_SE        | TRI_SE[0]            | D:          | COM15       | 07500200667370285916E645 | 0200            |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+

```
and the tests still failed to run:

```
Build successes:
  * TRI_SE::GCC_ARM::MBED-BUILD
  * TRI_SE::GCC_ARM::TESTS-STC-480-TESTCASE-1
mbedgt: greentea test automation tool ver. 1.2.0
mbedgt: test specification file 'C:\Projects\mbedos5\gl-apps\n4m-tests\.build/tests\TRI_SE\GCC_ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\Projects\mbedos5\gl-apps\n4m-tests\.build/tests\TRI_SE\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'TRI_SE' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.5959222841
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.14 sec
mbedgt: exited with code -110
[mbed] ERROR: "mbedgt" returned error code -110.
[mbed] ERROR: Command "mbedgt --test-spec C:\Projects\mbedos5\gl-apps\n4m-tests\.build/tests\TRI_SE\GCC_ARM\test_spec.json -n tests*" in "C:\Projects\mbedos5\gl-apps\n4m-tests"
---
```

when run mbedls -l it is listed mocked target:
```
...
| 0745             | NUCLEO_F303RE        |
| 0747             | NUCLEO_F303ZE        |
| 0750             | TRI_SE               |
| 0755             | NUCLEO_F070RB        |
| 0760             | NUCLEO_L073RZ        |
...
```

Would you advise what to try next?

mazimkhan, 2016-08-19:
```mbedls -m``` option creates a mock file in the current directory. Are running mbedgt from the same dir?

ivan-adzic-tridonic-com, 2016-08-19:
yes, after we have called mbedls -m option, we call:
```
$ mbed test -n tests* -m TRI_SE -t GCC_ARM
```
which is internally calling mbedgt.
Also as we have mentioned before it works fine if target is set as NUCLEO_F091. 

We do not know if we are calling mbed test does it change directory internally?

mazimkhan, 2016-08-21:
Can you please post your ```.build\tests\K64F\GCC_ARM\test_spec.json``` file, It will show the test binaries compiled by the build system.

ivan-adzic-tridonic-com, 2016-08-22:
Yes, this is test_spec.json
```
{
  "builds": {
    "TRI_SE-GCC_ARM": {
      "binary_type": "bootable", 
      "tests": {
        "tests-stc-480-testcase-1": {
          "binaries": [
            {
              "path": ".build/tests/TRI_SE/GCC_ARM/TESTS/stc-480/testcase-1/testcase-1.bin"
            }
          ]
        }
      }, 
      "toolchain": "GCC_ARM", 
      "base_path": ".build/tests/TRI_SE/GCC_ARM", 
      "baud_rate": 9600, 
      "platform": "TRI_SE"
    }
  }
}
```

also additional thing related to mbedls -m, mock was not created in current directory, it was created under Users in C:\Users\ivan.adzic\.mbed-ls\.mbedls-mock but it does not change anything if I move it to local directory.

mazimkhan, 2016-08-22:
Please try running greentea own its own with verbose logging as:

```
mbedgt -V --test-spec .build\tests\TRI_SE\GCC_ARM\test_spec.json
```

and share the output.

ivan-adzic-tridonic-com, 2016-08-22:
Hi, here it is:
```
$ mbedgt -V --test-spec .build/tests/TRI_SE/GCC_ARM/test_spec.json
mbedgt: greentea test automation tool ver. 1.2.0
mbedgt: test specification file '.build/tests/TRI_SE/GCC_ARM/test_spec.json' (specified with --test-spec option)
mbedgt: using '.build/tests/TRI_SE/GCC_ARM/test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_F091RC | NUCLEO_F091RC[0]     | COM15       | D:          | 07500200667370285916E645 |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'TRI_SE' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+-----------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id |
        +---------------+----------------------+-------------+-------------+-----------+
        +---------------+----------------------+-------------+-------------+-----------+
        use 0 instances of execution threads for testing
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.3581517649
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.09 sec
mbedgt: exited with code -110

```

mazimkhan, 2016-08-22:
Above ```NUCLEO_F091RC ``` indicates that the mock is not set. We have identified a bug in mbed-ls. When imported (like by greentea here) it is not including mocks. It has recently been introduced. Let me fix and release new version quickly for you.

mazimkhan, 2016-08-22:
Release mbed-ls version 1.2.1. Please try..

ivan-adzic-tridonic-com, 2016-08-22:
Thank you, it is working now! :+1: Board is picked up from mocks.

closed 2016-08-22 by ivan-adzic-tridonic-com

mazimkhan, 2016-08-22:
Great thanks for your patience too.