ARMmbed/greentea #181: Trying to create html report results in error if platform or toolchain contain "-"

cuevasj, 2016-08-24:
Using the command **mbedgt -V --report-html=my_report.html** resulted in the attached error.

I believe the root cause is that line 599 of mbed_report_api.py splits the string platform_toolchain without setting the optional maxsplit parameter to 1.  My toolchain is frdm-k64f-gcc so the result of splitting the string (on '-') is four substrings which cannot be assigned to two variables.

When I changed the line to platform, toolchain = platform_toolchain.split('-', 1), the error goes away and the html is generated.  I am very new to Python so maybe there's a better way to fix this.

[image]

[image]

bridadan, 2016-08-25:
@cuevasj Is this still and issue? I see that you also reported https://github.com/ARMmbed/greentea/issues/182 so it looks like you were able to create a report. Or was that using the patch you mentioned in your comment above?

cuevasj, 2016-08-25:
Yes, I generated the report using the patch described above.

PR #184 (2016-09-09): Fix for Issue 181
# Changes
* When splitting up <PLATFORM>-<TOOLCHAIN>, we now only take the last element of the split (the toolchain, as it should not contain a '-'), and then remove that from the original string to get the platform
* Problem occurred when platform contained a '-'

# Fixes
* This pull request fixes #181
closed 2016-09-09 by adbridge