ARMmbed/greentea #188: mbedgt: can test case sort be removed ?

jeromecoutant, 2016-09-07:
Hi

I would prefer that test case status is not sorted during mbedgt execution.

Ex:
Tested command: mbed test -t ARM -m NUCLEO_F070RB -vv -n tests-mbedmicro-rtos-mbed-threads

We can see the test suite execution :

[1473239592.78][CONN][RXD] >>> Running case #1: 'Testing single thread'...
[1473239592.98][CONN][RXD] >>> Running case #2: 'Testing parallel threads'...
[1473239593.18][CONN][RXD] >>> Running case #3: 'Testing serial threads'...

Then the test suite status:

mbedgt: test suite 'tests-mbedmicro-rtos-mbed-threads' ............................................... TIMEOUT in 50.15 sec
        test case: 'Testing parallel threads with wait' .............................................. OK in 0.17 sec
        test case: 'Testing single thread with wait' ................................................. OK in 0.15 sec
        test case: 'Testing serial threads with yield' ............................................... OK in 0.06 sec
        test case: 'Testing single thread with yield' ................................................ OK in 0.06 sec
        test case: 'Testing serial threads with wait' ................................................ OK in 1.06 sec
        test case: 'Testing single thread' ........................................................... OK in 0.06 sec
        test case: 'Testing parallel threads with child' ............................................. ERROR in 0.00 sec

==> Can this status can be displayed in the right order?

For report, I think we can keep the sort.

Thx

bridadan, 2016-09-12:
@jeromecoutant Just to be clear on what you're asking, you'd like the order of execution of tests to preserved in the test suite summary/status, but for the overall report it'd be ok to keep the alphabetical sorting?

jeromecoutant, 2016-09-12:
Hi
Yes, I think it is better for debug.

About overall report, it is up to you, but the alphabetical sort is good at this final step to comapre between several reports.

jeromecoutant, 2016-10-04:
Hi
Any update ? :-)

adbridge, 2016-10-04:
Hi Jerome ,
this is on our backlog of issues to address, however it's fairly low priority at the moment , so we can't give a definitive timescale on when it may be done I'm afraid. If you need a solution quicker then please feel free to submit a change and we'll see if we can get it merged in.