ARMmbed/greentea #191: TEST_ASSERT_MESSAGE message needs better user visibility

BlackstoneEngineering, 2016-09-12:
##Problem
The `TEST_ASSERT_MESSAGE(something,"Message about test case failing")` doesn't show up with appropriate panache. If a message is included here it is meant to only be seen if there is a failure. That means its important, which means it should draw the users attention somehow. The message is likely a diagnostic message from the test Author, or a useful tidbit to help the end user debug and pass the test. 

The problem is the 'message' is lost amongst a sea of crap. The problem is two fold, 
1) the message is hidden behind the `-v` flag, these messages only happen if there is a failure, which means it shouldnt be hidden behind the verbose flag, but instead shown front and center
2) the error message is hidden amongst all the other output text with no visible highlighting or color change on the terminal to highlight the message. 

Case in Point: Below is the failure output from a temperature test, the useful bit of text should say `Its too Hot (>10C), Faulty Sensor?`, try and find it.

Here is the test run normally
```
C:\cygwin\home\ausbla01\mbed\ci-test-shield>mbed test  -n tests-api-i2c
Building library mbed-build (K64F, GCC_ARM)
Building project GCC_ARM to tests-api-i2c (K64F, GCC_ARM)
Compile: I2C.cpp
[Warning] I2C.cpp@24: NULL used in arithmetic [-Wpointer-arith]
Link: tests-api-i2c
Elf2Bin: tests-api-i2c
+----------------------------+-------+-------+------+
| Module                     | .text | .data | .bss |
+----------------------------+-------+-------+------+
| Fill                       |    94 |     3 | 2472 |
| Misc                       | 38563 |  2224 | 2368 |
| core/hal                   | 19295 |    16 |  573 |
| core/rtos                  |  6831 |    24 | 2662 |
| features/FEATURE_IPV4      |    96 |     0 |   48 |
| frameworks/greentea-client |  1580 |    56 |   52 |
| frameworks/unity           |   584 |     0 |  160 |
| frameworks/utest           |  5312 |     5 |  805 |
| Subtotals                  | 72355 |  2328 | 9140 |
+----------------------------+-------+-------+------+
Allocated Heap: 65540 bytes
Allocated Stack: 32768 bytes
Total Static RAM memory (data + bss): 11468 bytes
Total RAM memory (data + bss + heap + stack): 109776 bytes
Total Flash memory (text + data + misc): 75723 bytes
Image: .build/tests/K64F/GCC_ARM/TESTS/API/I2C/tests-api-i2c.bin

Build successes:
  * K64F::GCC_ARM::MBED-BUILD
  * K64F::GCC_ARM::TESTS-API-I2C
mbedgt: greentea test automation tool ver. 1.2.0
mbedgt: test specification file 'C:\cygwin\home\ausbla01\mbed\ci-test-shield\.build/tests\K64F\GCC_ARM\test_spec.json' (specified with --test-spec opt
ion)
mbedgt: using 'C:\cygwin\home\ausbla01\mbed\ci-test-shield\.build/tests\K64F\GCC_ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: processing target 'K64F' toolchain 'GCC_ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: test case filter (specified with -n option)
        test filtered in 'tests-api-i2c'
mbedgt: running 1 test for platform 'K64F' and toolchain 'GCC_ARM'
mbedgt: mbed-host-test-runner: started
mbedgt: checking for GCOV data...
mbedgt: test on hardware with target id: 0240000025354e450025400d47d00046bb91000097969900
mbedgt: test suite 'tests-api-i2c' ................................................................... FAIL in 12.82 sec
        test case: 'Testing EEProm Read' ............................................................. OK in 0.05 sec
        test case: 'Testing EEProm Write' ............................................................ OK in 0.05 sec
        test case: 'Testing TMP102 Temperature Read' ................................................. FAIL in 0.11 sec
mbedgt: test case summary: 2 passes, 1 failure
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.1835863255
mbedgt: test suite report:
+--------------+---------------+---------------+--------+--------------------+-------------+
| target       | platform_name | test suite    | result | elapsed_time (sec) | copy_method |
+--------------+---------------+---------------+--------+--------------------+-------------+
| K64F-GCC_ARM | K64F          | tests-api-i2c | FAIL   | 12.82              | shell       |
+--------------+---------------+---------------+--------+--------------------+-------------+
mbedgt: test suite results: 1 FAIL
mbedgt: test case report:
+--------------+---------------+---------------+---------------------------------+--------+--------+--------+--------------------+
[... 189 lines cut ...]

## Solution
Bring the 'message' out from behind the `-v` flag and highlight it, either by placing it at the bottom of the log and marking it as `ERROR` or by highlighting it with red text in the output log. 

Ideally something like this:
```
mbedgt: test suite results: 1 FAIL
     Failure: "Its too Hot (>10C), Faulty Sensor?"
mbedgt: test case report:
+--------------+---------------+---------------+---------------------------------+--------+--------+--------+--------------------+
| target       | platform_name | test suite    | test case                       | passed | failed | result | elapsed_time (sec) | Message | 
+--------------+---------------+---------------+---------------------------------+--------+--------+--------+--------------------+
| K64F-GCC_ARM | K64F          | tests-api-i2c | Testing EEProm Read             | 1      | 0      | OK     | 0.05               | | 
| K64F-GCC_ARM | K64F          | tests-api-i2c | Testing EEProm Write            | 1      | 0      | OK     | 0.05               | | 
| K64F-GCC_ARM | K64F          | tests-api-i2c | Testing TMP102 Temperature Read | 0      | 1      | FAIL   | 0.11               | Its too Hot (>10C), Faulty Sensor?|
+--------------+---------------+---------------+---------------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 FAIL / 2 OK
        Failure: 'Its too Hot (>10C), Faulty Sensor?'
mbedgt: completed in 12.91 sec
mbedgt: exited with code 1
[mbed] ERROR: "mbedgt" returned error code 1.
[mbed] ERROR: Command "mbedgt --test-spec C:\cygwin\home\ausbla01\mbed\ci-test-shield\.build/tests\K64F\GCC_ARM\test_spec.json -n tests-api-i2c" in "C
:\cygwin\home\ausbla01\mbed\ci-test-shield"
---
```

BlackstoneEngineering, 2016-09-12:
@bridadan

adbridge, 2016-09-19:
ARM Internal Ref: IOTSFW-2923