ARMmbed/greentea #192: Memory map summary doesn't work any more

jeromecoutant, 2016-09-26:
Hi

All values are kept to 0

Ex:
Memory map breakdown for built projects (values in Bytes):
+----------------------------+---------------+-----------+------------+-------+------+-----------+-------------+
| name                       | target        | toolchain | static_ram | stack | heap | total_ram | total_flash |
+----------------------------+---------------+-----------+------------+-------+------+-----------+-------------+
| nist_internet_time_service | NUCLEO_F429ZI | ARM       |          0 |     0 |    0 |         0 |           0 |
| tcp_client_echo            | NUCLEO_F429ZI | ARM       |          0 |     0 |    0 |         0 |           0 |
| tcp_client_hello_world     | NUCLEO_F429ZI | ARM       |          0 |     0 |    0 |         0 |           0 |
| udp_echo_client            | NUCLEO_F429ZI | ARM       |          0 |     0 |    0 |         0 |           0 |
+----------------------------+---------------+-----------+------------+-------+------+-----------+-------------+

bridadan, 2016-09-26:
@jeromecoutant I believe this should be reported to https://github.com/ARMmbed/mbed-os, since that's where the memory map script lives. Would you mind moving this issue there?

jeromecoutant, 2016-09-27:
Hi
https://github.com/ARMmbed/mbed-os/issues/2820

closed 2016-09-27 by jeromecoutant