ARMmbed/greentea #196: Feature Request : Add ability to change UART Speed between board and DUT

BlackstoneEngineering, 2016-10-11:
## Problem
It is very hard to test the UART with greentea since greentea uses the UART line. Having the ability to change the speed of the channel for testing purposes would be very useful. 

## Solution
Implemeneted in htrun and greentea-client packages "key value protocol" can be decorated with few new simple commands like {{serial_speed;115200}} etc. + some simple HAL in greentea-client to change baudrate and restart serial if needed (+ flusing some buffers?)

 I can imagine on device and host side after baudrate change request there shjould be some simple recovery so both sides for sure are transmiting in new baudrate.
 So some implementation is in order in both greentea-client (C/C++) and htrun.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

BlackstoneEngineering, 2018-10-02:
This was filled out 2 years ago before there was any formalized anything, please specify what is missing.

closed 2018-11-05 by ciarmcom

BlackstoneEngineering, 2018-11-05:
@adbridge explain please?

adbridge, 2018-11-06:
It's been moved internally into a feature request by the looks of it. This one was thus closed by Earl's team.