ARMmbed/greentea #197: ERROR: "mbedgt: no platform/target matching tests were found!"

iotvietmember, 2016-10-12:
I have an error when using mbedgt.
I have a NRF51_DK board and I use Windows OS

1.Test with mbedls: mbedls --json
[image]
I think it is OK.

2. Test with mbedgt:
mbed-os$  mbed test --compile -m NRF51_DK -t GCC_ARM
mbed-os$  mbedgt -V -t NRF51_DK-GCC_ARM -n tests-integration-basic --test-spec BUILD\tests\NRF51_DK\GCC_ARM\test_spec.json
[image]

ERROR: "mbedgt: no platform/target matching tests were found!"

Why did I meet this problem?

adbridge, 2016-10-12:
The command line options you have used look strange.
Can you try mbedgt -l ?
This will list all the test cases greentea thinks are available .
You can also try removing "-t NRF51_DK-GCC_ARM" as that is the old Yotta way . See if that fixes your issue.

iotvietmember, 2016-10-12:
@adbridge Thank you. I tried removing "-t NRF51_DK-GCC_ARM" and mbedgt worked ok.
But my test met an error as the following pictures:
[image]
[image]

adbridge, 2016-10-12:
Looks like your device has just timed out. Try unplugging it from the usb connector and plugging it in and re-running. That can sometimes fix it.

iotvietmember, 2016-10-12:
I tried re-plug my device but that error is still appear

bridadan, 2016-10-12:
@iotvietmember It looks like your device isn't responding to the `{{__sync;xxxxx}}` command which is responsible for starting the test on the device.

To ensure that you can receive serial data correctly, I would recommend the following:

1. Open a connection to the board via PuTTy or another serial terminal (you can get the COM port from mbedls, baud rate should be 9600)
2. Flash the device with the test binary via drag-and-drop
3. Reset the device with the reset button after the flash process
4. Copy and paste the following text into the the terminal: `{{__sync;xxxxx}}`.

If you don't see anything appear in the serial terminal after that it looks like you're having a serial issue.
We can try and debug more after you've tried the above!

iotvietmember, 2016-10-12:
@bridadan  
As you said. I did not see anything apper in the terminal
I have configured the TX, RX pins accordance with hardware.

[image]

bridadan, 2016-10-12:
Are the pink lines the lines that you have entered? Or are those coming from the device? (Sorry, not familiar with your serial terminal)

iotvietmember, 2016-10-12:
@bridadan : Yes, The pink lines is the lines that I have entered

bridadan, 2016-10-12:
@iotvietmember Thanks for clarifying. Your device isn't responding, so this usually means there's an issue with the serial connection or your device has crashed.

Can you print from your device and see it on the serial port with just a simple application? Something like:

```
#include "mbed.h"

int main() {
    printf("This is a test\r\n");

    while(1) {
    }
}

iotvietmember, 2016-10-13:
@bridadan 
I am having a problem with mbed-os 5 and NRF51_DK board.

1. The following code is OK and led will blink
```
#include "mbed.h"

DigitalOut led1(p7);		
Serial pc(p10, p11); // tx, rx

// main() runs in its own thread in the OS
// (note the calls to Thread::wait below for delays)
int main() {
    pc.baud(9600);
    while (true) {
        led1 = !led1;
        Thread::wait(100);
        //pc.printf("Hello world");           /*Do not use this line*/
    }
}
```

2. The following code is NOT  OK and led will be turned on and never turn off.
```
#include "mbed.h"

DigitalOut led1(p7);		
Serial pc(p10, p11); // tx, rx

// main() runs in its own thread in the OS
// (note the calls to Thread::wait below for delays)
int main() {
    pc.baud(9600);
    while (true) {
        led1 = !led1;
        Thread::wait(100);
        pc.printf("Hello world");           /*Use this line*/
    }
}
```
I think that program crashes at that lines ("pc.printf("Hello world");")
PS: I build program with command "mbed compile -m NRF51_DK -t GCC_ARM"

bridadan, 2016-10-13:
@iotvietmember Your pin numbers appear to be incorrect for the USB serial lines: https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NORDIC/TARGET_NRF5/TARGET_MCU_NRF51822_UNIFIED/TARGET_NRF51_DK/PinNames.h#L139

You can see from here that the `USBTX` pin is `p9` and the `USBRX` pin is `p11`. If you remove the `Serial` object and just use the standard `printf` you should be able to see serial data. You may also need to add a `\n` to your printf to ensure the buffer is flushed.

```
#include "mbed.h"

DigitalOut led1(p7);

// main() runs in its own thread in the OS
// (note the calls to Thread::wait below for delays)
int main() {
    while (true) {
        led1 = !led1;
        Thread::wait(100);
        printf("Hello world\r\n");
    }
}
```

iotvietmember, 2016-10-13:
@bridadan 
I am using a customize board with nRF51822xxAC chip. On my board, TX pin is p10, RX pin is p11.
I tried to use mbed Online Compiler to build the following code and it worked well on my board.
```
#include "mbed.h"
 
Serial pc(p10, p11); // tx, rx
int main() {
    pc.baud(9600);
    while (true) {
        pc.printf("Hello world\r\n");        
    }
}
```

iotvietmember, 2016-10-13:
I think that the NRF51_DK board has not worked well at mbed-OS5 ????

bridadan, 2016-10-13:
The NRF51_DK board definitely works with mbed OS 5, we run tests on it every night. I'm afraid I'm not sure what's going wrong though with your custom board. What kind of interface chip are you using?

iotvietmember, 2016-10-13:
@bridadan  I am using a DAPLink (version 0242) interface with LPC11U35 chip. 

https://cloud.githubusercontent.com/assets/15735613/19304244/0a03b5f6-9096-11e6-8c6d-f2bfb58c7556.PNG

bridadan, 2016-10-13:
@iotvietmember Thanks for the info. Unfortunately it's hard to say what's actually going wrong without having a board. I would try stepping through the code in a debugger to see where it's going wrong if you can.

iotvietmember, 2016-10-13:
@bridadan  Do you please to guide me how to generate my code to GNU ARM toolchain (GCC_ARM compiler)? 
For example, GNU ARM Eclipse. I tried debug by OpenOCD and it worked well

bridadan, 2016-10-13:
I've never done it myself, so I won't be much of a guide unfortunately. I've used Keil uVision in the past, but that will most likely require a license for mbed OS code (due to code size). There may be some guides on developer.mbed.org, I'd try searching there.

iotvietmember, 2016-10-13:
@bridadan  I tried to export my program to Keil MDK for debug application.
```mbed export -m NRF51_DK -i uvision```
Unfortunately, I met a lot of errors when I build project.

[image]

bridadan, 2016-10-13:
Is your version of Keil uVision up to date?

iotvietmember, 2016-10-13:
[image]

This is information about Keil uVision that I have used

bridadan, 2016-10-13:
I think the issues you are seeing is usually due to having an older version of Armcc.exe. If you update your version of uVision those issues should go away I believe.

iotvietmember, 2016-10-14:
@bridadan I am using GNU ARM Eclipse to debug my project. I created a breakpoint at command "pc.printf......" and I found that my program is crash at Mutex Service Calls declarations in rt_CMSIS.c file.

[image]

iotvietmember, 2016-10-14:
[image]
The above picture is source code of nrf_uart_event_check

iotvietmember, 2016-10-17:
@bridadan Can you help me fix this error?

bridadan, 2016-10-17:
@iotvietmember This appears to be a device software issue, not a greentea or testing tool issue. I'm afraid I'm not sure exactly what the issue is. You may have better luck asking this question on the developer site (developer.mbed.org) as many people have created boards based on this target.

iotvietmember, 2016-10-18:
@bridadan  Thanks for your help!

closed 2016-10-19 by iotvietmember