ARMmbed/greentea #198: Python error while trying to test NUCLEO_F429ZI and latest mbed-greentea

screamerbg, 2016-10-14:
### Platform: Windows
```
>mbedgt --version
1.2.2
>mbed --version
0.9.9
>mbedls --version
1.2.8
```

Running `mbed test -t ARM -m NUCLEO_F429ZI`

```
Build successes:
  * NUCLEO_F429ZI::ARM::FEATURES-FEATURE_LWIP-TESTS-MBEDMICRO-NET-NIST_INTERNET_TIME_SERVICE
  * NUCLEO_F429ZI::ARM::FEATURES-FEATURE_LWIP-TESTS-MBEDMICRO-NET-TCP_CLIENT_ECHO
  * NUCLEO_F429ZI::ARM::FEATURES-FEATURE_LWIP-TESTS-MBEDMICRO-NET-TCP_CLIENT_HELLO_WORLD
  * NUCLEO_F429ZI::ARM::FEATURES-FEATURE_LWIP-TESTS-MBEDMICRO-NET-UDP_ECHO_CLIENT
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-BASIC_TEST
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-BASIC_TEST_DEFAULT
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_ASYNC_VALIDATE
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_CONTROL_ASYNC
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_CONTROL_REPEAT
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_SELECTION
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_SETUP_FAILURE
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CASE_TEARDOWN_FAILURE
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-CONTROL_TYPE
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-MINIMAL_ASYNC_SCHEDULER
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-MINIMAL_SCHEDULER
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_ASSERTION_FAILURE_TEST_SETUP
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_SETUP_CASE_SELECTION_FAILURE
  * NUCLEO_F429ZI::ARM::FEATURES-FRAMEWORKS-UTEST-TESTS-UNIT_TESTS-TEST_SETUP_FAILURE
  * NUCLEO_F429ZI::ARM::MBED-BUILD
  * NUCLEO_F429ZI::ARM::RTOS-RTX-TARGET_CORTEX_M-TESTS-MEMORY-HEAP_AND_STACK
  * NUCLEO_F429ZI::ARM::TESTS-EVENTS-QUEUE
  * NUCLEO_F429ZI::ARM::TESTS-INTEGRATION-BASIC
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-MBED-ATTRIBUTES
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-MBED-CALL_BEFORE_MAIN
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-MBED-CPP
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-MBED-DIV
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-BASIC
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-ISR
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-MAIL
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-MALLOC
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-MUTEX
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-QUEUE
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-SEMAPHORE
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-SIGNALS
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-THREADS
  * NUCLEO_F429ZI::ARM::TESTS-MBEDMICRO-RTOS-MBED-TIMER
  * NUCLEO_F429ZI::ARM::TESTS-MBEDTLS-SELFTEST
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-C_STRINGS
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-DEV_NULL
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-ECHO
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-GENERIC_TESTS
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-LP_TIMEOUT
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-RACE_TEST
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-RTC
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-STL_FEATURES
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-TICKER
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-TIMEOUT
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-WAIT_US
  * NUCLEO_F429ZI::ARM::TESTS-MBED_FUNCTIONAL-CALLBACK
  * NUCLEO_F429ZI::ARM::TESTS-MBED_FUNCTIONAL-CALLBACK_BIG
  * NUCLEO_F429ZI::ARM::TESTS-MBED_FUNCTIONAL-CALLBACK_SMALL
  * NUCLEO_F429ZI::ARM::TESTS-MBED_FUNCTIONAL-FUNCTIONPOINTER
  * NUCLEO_F429ZI::ARM::TESTS-MBED_HAL-LP_TICKER

Build skips:
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-MEM_TRACE
  * NUCLEO_F429ZI::ARM::TESTS-MBED_DRIVERS-STATS
  * NUCLEO_F429ZI::ARM::TESTS-STORAGE_ABSTRACTION-BASICAPI
mbedgt: greentea test automation tool ver. 1.2.2
mbedgt: test specification file 'C:\mbed\mbed-os\BUILD\tests\NUCLEO_F429ZI\ARM\test_spec.json' (specified with --test-spec option)
mbedgt: using 'C:\mbed\mbed-os\BUILD\tests\NUCLEO_F429ZI\ARM\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 1 device
mbedgt: mbed-ls was unable to enumerate correctly all properties of the device!
        check with 'mbedls -j' command if all properties of your device are enumerated properly
mbedgt: mbed-ls property 'serial_port' is 'None'
        check if your serial port driver is correctly installed!
mbedgt: processing target 'NUCLEO_F429ZI' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: unexpected error:
        'NoneType' object has no attribute 'endswith'
Traceback (most recent call last):
  File "C:\Program Files\Python\Scripts\mbedgt-script.py", line 9, in <module>
    load_entry_point('mbed-greentea==1.2.2', 'console_scripts', 'mbedgt')()
  File "C:\Progra~1\Python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 389, in main
    cli_ret = main_cli(opts, args)
  File "C:\Progra~1\Python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 784, in main_cli
    if not mbed_dev['serial_port'].endswith(str(baudrate)):
AttributeError: 'NoneType' object has no attribute 'endswith'
[mbed] ERROR: "mbedgt" returned error code 1.
[mbed] ERROR: Command "mbedgt --test-spec C:\mbed\mbed-os\BUILD\tests\NUCLEO_F429ZI\ARM\test_spec.json" in "C:\mbed\mbed-os"
---
```

mazimkhan, 2016-10-14:
Can you please run ```mbedls``` and post the output. It looks like serial port is not detected.

adbridge, 2016-10-14:
Looks more like an mbed ls issue will point @mazimkhan here

screamerbg, 2016-10-14:
Also
```
>mbed detect

[mbed] Detected NUCLEO_F429ZI, port COM73, mounted E:
[mbed] Supported toolchains for NUCLEO_F429ZI
+---------------+-----------+-----------+-----------+-----------+-----------+
| Target        | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |
+---------------+-----------+-----------+-----------+-----------+-----------+
| NUCLEO_F429ZI | Supported | Supported | Supported | Supported | Supported |
+---------------+-----------+-----------+-----------+-----------+-----------+
Supported targets: 1
Supported toolchains: 3
```

And

```
>mbedls
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
| NUCLEO_F429ZI | NUCLEO_F429ZI[0]     | E:          | COM73       | 07960221002962073F6DF4AC | 0221            |
+---------------+----------------------+-------------+-------------+--------------------------+-----------------+
```

screamerbg, 2016-10-14:
cc @mazimkhan @adbridge

screamerbg, 2016-10-14:
It worked just fine for ST Nucleo F401, but not for ST Nucleo F429ZI

screamerbg, 2016-10-14:
Seems to be related to the com-port or something being busy between calls. Will close the ticket for now as I'm not able to reproduce it currently. Perhaps the code should handle a case when there is a platform but for some reason the com port is not available (which seems to be valid scenario given that I got the python exception).

`mbed_greentea\mbed_greentea_cli.py, Line 780`

```

            if mbed_dev['platform_name'] == platform_name:
                # We will force configuration specific baudrate by adding baudrate to serial port
                # Only add baudrate decoration for serial port if it's not already there
                # Format used by mbedhtrun: 'serial_port' = '<serial_port_name>:<baudrate>'
                if not mbed_dev['serial_port'].endswith(str(baudrate)):
```

screamerbg, 2016-10-14:
Just managed to reproduce it. Happens when Windows has installed everything but the serial comport.

```
        +-------------------+----------------------+-------------+-------------+--------------------------------------------------+
        | platform_name     | platform_name_unique | serial_port | mount_point | target_id                                        |
        +-------------------+----------------------+-------------+-------------+--------------------------------------------------+
        | K64F              | K64F[0]              | COM74       | E:          | 0240000033514e45003e500585d4002fe981000097969900 |
        | UBLOX_EVK_ODIN_W2 | UBLOX_EVK_ODIN_W2[0] | None        | H:          | 12360200000000003F65830B                         |
        +-------------------+----------------------+-------------+-------------+--------------------------------------------------+
mbedgt: processing target 'UBLOX_EVK_ODIN_W2' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
mbedgt: unexpected error:
        'NoneType' object has no attribute 'endswith'
Traceback (most recent call last):
  File "C:\Program Files\Python\Scripts\mbedgt-script.py", line 9, in <module>
    load_entry_point('mbed-greentea==1.2.2', 'console_scripts', 'mbedgt')()
  File "C:\Progra~1\Python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 389, in main
    cli_ret = main_cli(opts, args)
  File "C:\Progra~1\Python\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 784, in main_cli
    if not mbed_dev['serial_port'].endswith(str(baudrate)):
AttributeError: 'NoneType' object has no attribute 'endswith'
```

Think this should be handled in greenea not as a warning, but error.

adbridge, 2016-11-01:
Looking at the code, I think we should print out an error stating that the serial port was not detected correctly and then skip that MUT. 
So the problem code is :  
```
if not mbed_dev['serial_port'].endswith(str(baudrate)):
    mbed_dev['serial_port'] = "%s:%d" % (mbed_dev['serial_port'], baudrate)
```
So we want something like:
```
sp = mbed_dev['serial_port']
if sp and not sp.endswith(str(baudrate)):
    mbed_dev['serial_port'] = "%s:%d" % (mbed_dev['serial_port'], baudrate)
else:
    gt_logger.gt_log_err("Serial port for target %s not detected correctly\n" % mbed_dev['target_id'])
```

adbridge, 2016-11-01:
@mazimkhan this look reasonable to you?

mazimkhan, 2016-11-01:
Yes it would be more clear then the exception.

closed 2016-11-02 by adbridge