ARMmbed/greentea #202: Retargeting does not work through mbedgt

stevew817, 2016-11-04:
When retargeting a platform using the retarget functionality in mbed-ls, the retarget shows up when executing mbedls, but not when running mbedgt.
Does mbedgt call mbedls from a different working directory?

bridadan, 2016-11-04:
Hi @stevew817, could please post the output of `pip list`? I believe this issue was patched in [mbed-ls version 1.2.1](https://github.com/ARMmbed/mbed-ls/releases/tag/v1.2.1). If you find you're using an earlier verison of mbed-ls, please run `pip install -U mbed-ls` to see if that solves your issue. Thanks!

stevew817, 2016-11-04:
I did upgrade to latest...

pip-list:

> mbed-cli (0.9.10)
mbed-greentea (1.2.3)
mbed-host-tests (1.1.4)
mbed-ls (1.2.9)
mbed-test-wrapper (1.0.0)
mock (2.0.0)
monotonic (1.2)

stevew817, 2016-11-04:
I think the issue is that most tools (greentea, htrun, ...) call mbedls in the following way:

> mbeds = mbed_lstools.create()
    mbeds_list = mbeds.list_mbeds_ext()

I discovered you need to specifically call retarget() on mbeds for retarget to work... So maybe the solution here is to call retarget in the create function?

stevew817, 2016-11-04:
@bridadan The issue you referred to was for mock targets, mine is for target redirection...

bridadan, 2016-11-04:
Oops, you're definitely right, sorry for the hasty reply!

> So maybe the solution here is to call retarget in the create function? 

I think you're right @stevew817, that seems like a good solution to me. Would you mind trying it and if it fixes your issue, submit a patch to mbed-ls?

stevew817, 2016-11-07:
@bridadan Fix PR submitted.

bridadan, 2016-12-13:
@stevew817 Your fix has been released in the latest version of mbedls, would you mind confirming that this fixes this issue?

bridadan, 2017-06-06:
@mazimkhan  I think that this issue is safe to close

closed 2017-06-07 by mazimkhan