ARMmbed/greentea #205: Greentea reports success even though a test failed

c1728p9, 2016-12-23:
Greentea is ignoring a test failure and reporting success. Log of this shown here:
[consoleText.txt](https://github.com/ARMmbed/greentea/files/670178/consoleText.txt)

adbridge, 2016-12-23:
Looks like one test case returned IOERR_SERIAL , which means there was a problem communicating with the board at that point. I suspect that is not considered a 'test failure' . Do you get the same error if you run it again ? It is most likely a transient hardware issue by the looks of it.

bridadan, 2016-12-27:
After looking into this more, I'm not sure this is a bug in Greentea. The Jenkins job that was running greentea was ignoring the return value of the process and instead using the Junit report to determine if there was a failure. The IOERR_SERIAL error is not in the Junit report, so it didn't cause the job to fail. The CI should really use the return code from greentea to determine if the process was successful, so I've since changed this in the Jenkins setup.

The reason the IOERR_SERIAL isn't in the Junit report is because it happened between test cases. So you can't really associate the serial error with any particular test case, unless you want to invalidate all the rest of the tests, or maybe just add it to the last good test case for that binary? Honestly not sure. It currently shows up as an error if the serial error occurs in the middle of a test case, as expected.

**So TL;DR;** Serial error occurred exactly at the right time and it wasn't included in the Junit report. Do we need to specify the behavior of this case so that the error gets included? Or do we leave it as is (since technically no test case actually failed)?

adbridge, 2016-12-30:
This sounds like one of those weird, infrequent corner cases. I wonder if we could get Jenkins to re-run the binary if Greentea returns this kind of status?

bridadan, 2017-01-03:
@adbridge I think in this case realizing greentea is returning an error code may be enough. I honestly can't think of a good place to put the error, unless we want to just mark the last successful test case as an error? Which is then kinda lame because that test case actually executed to completion.

theotherjimmy, 2017-09-20:
@bridadan @c1728p9 @adbridge Have you come to a conclusion on how we should handle this corner case?

adbridge, 2017-09-21:
Is it possible to have more than one return value ? Ie something like 
test case status - Pass/Fail
h/w status - Stable/Faulty 

Something like that? I guess we could encode those into one returned value that could be parsed ?

jupe, 2018-06-06:
could it be pass/fail/inconclusive, where inconc means failures that doesn’t related to SUT (software under test) - so hw/test script/test framework failures etc. If possible there could be another property - reason which could describe bit more what went wrong.. This is widely used approach generally..

aashishc1988, 2018-07-23:
[TestResultsWithIOERR.txt](https://github.com/ARMmbed/greentea/files/2220645/TestResultsWithIOERR.txt)
Please see the attached test results. I tried replicating the behavior on my system, and now if an IOERR_SERIAL error occurs, we do report it as a failure in test results. The greentea version listed on the issue is 1.2.4, and we are on 1.4.0 now, so its possible it was fixed somewhere along the way.

aashishc1988, 2018-07-23:
How I tried to reproduce the issue:
Open a comm port to the part, and on another window send command,

> mbedgt -m NUCLEO_F103RB -n features-tests-filesystem-heap_block_device,tests-mbed_drivers-c_strings

to run some tests on the part. I cannot reproduce undesired behavior (not reporting ioerr as an error) anymore, because now we report it as an error with greentea version 1.4.0. 
I am going to close the linked JIRA.

@theotherjimmy  I do not have permissions to close the issue on github. So someone else might have to do that.

closed 2018-07-23 by theotherjimmy