ARMmbed/greentea #208: detection of host test not consistent when using the -n option

LiyouZhou, 2017-01-23:
when I use the -n "tests-unit-unittest"
the host test can be detected in "TESTS/unit/host_tests"

But if I don't use the -n option
the host test have to reside in "TESTS/host_tests" in order to be detected.

bridadan, 2017-01-26:
Thanks for reporting @LiyouZhou. I have a feeling you're right, but I haven't had a chance to investigate this further. I believe this is the function that's responsible for detecting host tests: https://github.com/ARMmbed/greentea/blob/6cb3e595e0e61857da69b01a0edaa231b45481df/mbed_greentea/mbed_test_api.py#L160

If you're interested I'd start looking there (it may be a fairly simple fix), otherwise myself or @mazimkhan will have to take a look.

LiyouZhou, 2017-01-26:
related: https://github.com/ARMmbed/greentea/issues/210 that detection function is badly implemented.