ARMmbed/greentea #210: host test detection does not support custom build directories

LiyouZhou, 2017-01-25:
if I compile program with mbed test --build <build-dir> option, the host test detection will fail.

This method used to detect host test direction is not robust https://github.com/ARMmbed/greentea/blob/master/mbed_greentea/mbed_test_api.py#L160

In theory, the test_spec.json already give a "base_path" field to tell mbedgt what the base build directory is. This would be a more robust way to detecting relative path of host tests.

theotherjimmy, 2017-09-20:
I take it that this is still an issue. Is that correct @LiyouZhou?

LiyouZhou, 2017-09-21:
I think so.

theotherjimmy, 2018-10-02:
Or we could specify the host tests directory, that would be even more robust, considering that the build system has to know to ignore it.

Alex-EEE, 2018-10-03:
@theotherjimmy, this is similar to https://github.com/ARMmbed/htrun/issues/217  This issue is actually why I started playing around with -e.  As far as I can tell, host test detection doesn't work.  The docs say that they should be picked up if the host test is inside TESTS/host_tests ( or something like that ), but it never worked for me, maybe for the same reason we were discussing on 217 ( windows / python global variable thing )

I would prefer specifying a directory, or list of directories.  I don't really mess with the test_spec.json ( mbed CLI creates that, right ? ), so I would prefer command line over messing with a json file

bridadan, 2019-01-07:
@theotherjimmy Did you fix this in https://github.com/ARMmbed/htrun/pull/219?