ARMmbed/greentea #215: --digest functionality on longer works

LiyouZhou, 2017-03-08:
```
echo "sth" | mbedgt --digest stdin -v -V
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: using 'module.json' from current directory!
mbedgt: checking for yotta target in current directory
	reason: no --target switch set
mbedgt: parsing local file '.yotta.json' for target information
mbedgt: assuming default target as 'custom-linux-x86-gcc'
mbedgt: No keyword with format "mbed-target:<platform name>" found in target.json!
Can not determine required mbed platform name!
mbedgt: No targets.json files found, using default target properties
mbedgt: unexpected error:
	run_host_test() got an unexpected keyword argument 'hooks'
Traceback (most recent call last):
  File "/usr/local/bin/mbedgt", line 11, in <module>
    load_entry_point('mbed-greentea==1.2.5', 'console_scripts', 'mbedgt')()
  File "/usr/local/lib/python2.7/dist-packages/mbed_greentea/mbed_greentea_cli.py", line 389, in main
    cli_ret = main_cli(opts, args)
  File "/usr/local/lib/python2.7/dist-packages/mbed_greentea/mbed_greentea_cli.py", line 684, in main_cli
    verbose=verbose)
TypeError: run_host_test() got an unexpected keyword argument 'hooks'
```

theotherjimmy, 2017-09-20:
@LiyouZhou I can't seem to reproduce this issue. Does this command have to be run from a particular directory?

LiyouZhou, 2017-09-21:
Just updated to 1.2.6, this seem to have improved.

theotherjimmy, 2017-09-21:
> this seem to have improved.

Is the bug still present?

LiyouZhou, 2017-09-21:
doesn't seem like I will close for now.

closed 2017-09-21 by LiyouZhou