ARMmbed/greentea #216: Caps lock support for test -n xxxx-xxxxx-xxxx

BlackstoneEngineering, 2017-03-17:
Original Issue raised here : https://github.com/ARMmbed/mbed-cli/issues/454#issuecomment-287376434

## Problem
When using the `test -n` command the test name must be input in lower case, even if the folders represented by the test name are all in caps. This has lead to a great deal of confusion for developers. 

### example
CI Test shield test for SPI is located in the folder `TESTS/API/SPI`, so to run this test you would use `test -n tests-api-spi`. I have observed multiple developers try to use `test -n TESTS-API-SPI` because the name of the folders are in caps, so they understand the folder names seperated by dashes, but the lower case conversion almost always throws them.

## Proposal
To make the tool more intuitive I propose accepting either caps lock or lower case and then munging it as appropriate before passing it on. That way the tool becomes more intuitive to use.

theotherjimmy, 2018-06-08:
I love the name.

theotherjimmy, 2018-06-08:
I need an "ergonomics" label.