ARMmbed/greentea #220: mbed_drivers "c_strings" test case is failing in greenttea test.

pranavshandilya, 2017-04-18:
When I added "default_lib": "small", in targets.json under LPC1768 defination.
snip---
    "LPC1768": {
        "inherits": ["LPCTarget"],
        "core": "Cortex-M3",
        "extra_labels": ["NXP", "LPC176X", "MBED_LPC1768"],
        "supported_toolchains": ["ARM", "uARM", "GCC_ARM", "GCC_CR", "IAR"],
        "detect_code": ["1010"],
        "device_has": ["ANALOGIN", "ANALOGOUT", "CAN", "DEBUG_AWARENESS", "ERROR_PATTERN", 
        "ETHERNET", "I2C", "I2CSLAVE", "INTERRUPTIN", "LOCALFILESYSTEM", "PORTIN", "PORTINOUT", 
        "PORTOUT", "PWMOUT", "RTC", "SEMIHOST", "SERIAL", "SERIAL_FC", "SLEEP", "SPI", "SPISLAVE", 
        "STDIO_MESSAGES"],
        **"default_lib": "small",**
        "release_versions": ["2", "5"],
        "features": ["LWIP"],
        "device_name": "LPC1768"
    },
-----
Inside c_strings test suits below test case are faling.

snip----
void test_case_c_string_f_f() {
    CLEAN_BUFFER;
    sprintf(buffer, "%f %f %f %f %f %f %f %f %f %f", FLOATS);
    TEST_ASSERT_EQUAL_STRING("0.002000 0.924300 15.913200 791.773680 6208.200000 25719.495200 426815.982588 6429271.046000 42468024.930000 212006462.910000", buffer);
}
-----

I just want to check why "default_lib": "small", is used in other controllers, not in LPC1768.
After adding this option to LPC1768, I realized that GCC_ARM RTOS test cases(single test) are not supported in LPC1768 controller.
Why this?

And why only one test case is failing in c_strings test suits?

bridadan, 2017-04-18:
A similar question was actually posted to the developer site today, please see the post for more details: https://developer.mbed.org/questions/77776/greentea-test-race_test-is-failing-in-GC/

pranavshandilya, 2017-04-19:
@bridadan oh ok, but no one replied there.
Please tell me the solution for this issue?

bridadan, 2017-04-19:
Also just realized I linked to the wrong question :smile: : https://developer.mbed.org/questions/77773/Greentea-test-tests-mbed_drivers-c_strin/

That one has more information in it. Would you mind closing this issue if you feel it is resolved?

closed 2017-04-19 by pranavshandilya