ARMmbed/greentea #221: Parallel test

jeromecoutant, 2017-04-27:
Hi

It seems that we can't make several MBED test in parallel...
Is support of parallel tests planned ?

Thx

bridadan, 2017-04-27:
@jeromecoutant Are you using the `--parallel` flag in greentea or are you just running multiple instance of `mbed test`/`mbedgt`?

jeromecoutant, 2017-04-27:
Hi

I tried both:

--parallel doesn't work to make parallel test, I didn't check, but maybe it is more linked to the parallel compilation (-j option ?)

I also tried to start 2 mbedgt commands in 2 different consoles (2 different targets). The second test goes to TIMEOUT as it couldn't sync I think.

bridadan, 2017-04-27:
I would expect there may be issues with `--parallel`, so I'd avoid that for now. It's not a very often used/tested feature unfortunately.

However I do know that 2 `mbedgt` commands should work just fine as we do this every day.

Are the targets different models or the same model? For instance, if they were both NUCLEO_F411RE boards, you would need to tell greentea which target ID to use for each instance of `mbedgt`. This can be done with the [--use-tids option](https://github.com/ARMmbed/greentea#selecting-platforms).

If you're already doing all of that and it's still not working, can you please post the verbose logs from the test run of both boards?

jeromecoutant, 2017-04-27:
Ex:
mbed test -t ARM -m NUCLEO_F446ZE -n tests-mbed_drivers-echo -vv --compile
...

mbed test -t ARM -m NUCLEO_L476RG -n tests-mbed_drivers-echo -vv --compile
...

mbedgt -V -v -t NUCLEO_F446ZE-ARM,NUCLEO_L476RG-ARM --parallel=2
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: using multiple test specifications from current directory!
        using 'BUILD\tests\NUCLEO_F446ZE\ARM\test_spec.json'
        using 'BUILD\tests\NUCLEO_L476RG\ARM\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 3 devices
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_L476RG | NUCLEO_L476RG[0]     | COM8        | D:          | 07650221625265605D16F338 |
        | NUCLEO_F103RB | NUCLEO_F103RB[0]     | COM5        | E:          | 07000221C939931AF67D0527 |
        | NUCLEO_F446ZE | NUCLEO_F446ZE[0]     | COM9        | F:          | 07780221054667783A02F13D |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'NUCLEO_L476RG' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 2)
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_L476RG | NUCLEO_L476RG[0]     | COM8:9600   | D:          | 07650221625265605D16F338 |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: running 1 test for platform 'NUCLEO_L476RG' and toolchain 'ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m NUCLEO_L476RG -p COM8:9600 -f "BUILD/tests/NUCLEO_L476RG/ARM/TESTS/mbed_drivers/echo/echo.bin" -e "TESTS\host_tests" -d D: -C 4 -c shell -t 07650221625265605D16F338
mbedgt: mbed-host-test-runner: started
[1493306090.55][HTST][INF] host test executor ver. 1.1.7
[1493306090.55][HTST][INF] copy image onto target...
...
mbedgt: all tests finished!
mbedgt: processing target 'NUCLEO_F446ZE' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 2)
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_F446ZE | NUCLEO_F446ZE[0]     | COM9:9600   | F:          | 07780221054667783A02F13D |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: running 1 test for platform 'NUCLEO_F446ZE' and toolchain 'ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m NUCLEO_F446ZE -p COM9:9600 -f "BUILD/tests/NUCLEO_F446ZE/ARM/TESTS/mbed_drivers/echo/echo.bin" -e "TESTS\host_tests" -d F: -C 4 -c shell -t 07780221054667783A02F13D
mbedgt: mbed-host-test-runner: started
[1493306121.60][HTST][INF] host test executor ver. 1.1.7
[1493306121.60][HTST][INF] copy image onto target...
[1493306121.60][COPY][INF] Waiting up to 60 sec for '07780221054667783A02F13D' mount point (current is 'F:')...
        1 file(s) copied.
[1493306128.74][HTST][INF] starting host test process...
...

==> targets have been executed 1 by 1

jeromecoutant, 2017-04-27:
2nd test:

In one console window:

$ mbedgt -V -v -t NUCLEO_L476RG-ARM
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: using multiple test specifications from current directory!
        using 'BUILD\tests\NUCLEO_F446ZE\ARM\test_spec.json'
        using 'BUILD\tests\NUCLEO_L476RG\ARM\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 3 devices
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_L476RG | NUCLEO_L476RG[0]     | COM8        | D:          | 07650221625265605D16F338 |
        | NUCLEO_F103RB | NUCLEO_F103RB[0]     | COM5        | E:          | 07000221C939931AF67D0527 |
        | NUCLEO_F446ZE | NUCLEO_F446ZE[0]     | COM9        | F:          | 07780221054667783A02F13D |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'NUCLEO_L476RG' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_L476RG | NUCLEO_L476RG[0]     | COM8:9600   | D:          | 07650221625265605D16F338 |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: running 1 test for platform 'NUCLEO_L476RG' and toolchain 'ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m NUCLEO_L476RG -p COM8:9600 -f "BUILD/tests/NUCLEO_L476RG/ARM/TESTS/mbed_drivers/echo/echo.bin" -e "TESTS\host_tests" -d D: -C 4 -c shell -t 07650221625265605D16F338
mbedgt: mbed-host-test-runner: started
...
mbedgt: completed in 36.99 sec

In an other console window (same time):

$ mbedgt -V -v -t NUCLEO_F446ZE-ARM
mbedgt: greentea test automation tool ver. 1.2.5
mbedgt: using multiple test specifications from current directory!
        using 'BUILD\tests\NUCLEO_F446ZE\ARM\test_spec.json'
        using 'BUILD\tests\NUCLEO_L476RG\ARM\test_spec.json'
mbedgt: detecting connected mbed-enabled devices...
mbedgt: detected 3 devices
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_L476RG | NUCLEO_L476RG[0]     | COM8        | D:          | 07650221625265605D16F338 |
        | NUCLEO_F103RB | NUCLEO_F103RB[0]     | COM5        | E:          | 07000221C939931AF67D0527 |
        | NUCLEO_F446ZE | NUCLEO_F446ZE[0]     | COM9        | F:          | 07780221054667783A02F13D |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: processing target 'NUCLEO_F446ZE' toolchain 'ARM' compatible platforms... (note: switch set to --parallel 1)
        +---------------+----------------------+-------------+-------------+--------------------------+
        | platform_name | platform_name_unique | serial_port | mount_point | target_id                |
        +---------------+----------------------+-------------+-------------+--------------------------+
        | NUCLEO_F446ZE | NUCLEO_F446ZE[0]     | COM9:9600   | F:          | 07780221054667783A02F13D |
        +---------------+----------------------+-------------+-------------+--------------------------+
mbedgt: running 1 test for platform 'NUCLEO_F446ZE' and toolchain 'ARM'
        use 1 instance of execution threads for testing
mbedgt: checking for 'host_tests' directory above image directory structure
        found 'host_tests' directory in: 'TESTS\host_tests'
mbedgt: selecting test case observer...
        calling mbedhtrun: mbedhtrun -m NUCLEO_F446ZE -p COM9:9600 -f "BUILD/tests/NUCLEO_F446ZE/ARM/TESTS/mbed_drivers/echo/echo.bin" -e "TESTS\host_tests" -d F: -C 4 -c shell -t 07780221054667783A02F13D
mbedgt: mbed-host-test-runner: started
[1493306334.85][HTST][INF] host test executor ver. 1.1.7
[1493306334.85][HTST][INF] copy image onto target...
[1493306334.85][COPY][INF] Waiting up to 60 sec for '07780221054667783A02F13D' mount point (current is 'F:')...
        1 file(s) copied.
[1493306344.46][HTST][INF] starting host test process...
[1493306345.54][CONN][INF] starting connection process...
[1493306345.54][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[... 26 lines cut ...]
        no test case report present, assuming test suite to be a single test case!
        test suite: tests-mbed_drivers-echo
        test case: tests-mbed_drivers-echo
mbedgt: test on hardware with target id: 07780221054667783A02F13D
mbedgt: test suite 'tests-mbed_drivers-echo' ......................................................... TIMEOUT in 76.70 sec
[... cut ...]
+-------------------+---------------+-------------------------+-------------------------+--------+--------+--------+--------------------+
| NUCLEO_F446ZE-ARM | NUCLEO_F446ZE | tests-mbed_drivers-echo | tests-mbed_drivers-echo | 0      | 1      | ERROR  | 76.7               |
+-------------------+---------------+-------------------------+-------------------------+--------+--------+--------+--------------------+
mbedgt: test case results: 1 ERROR
mbedgt: completed in 86.01 sec
mbedgt: exited with code 1

bridadan, 2017-04-27:
Great, thanks for the logs. Can you test the `NUCLEO_F446ZE` by itself just fine? Does it fail only when being ran with the `NUCLEO_L476RG`?

jeromecoutant, 2017-06-15:
Hi
Any update about this parallel issue ?
Are you using this parallel feature on your side ?
Thx

bridadan, 2017-06-16:
Hi @jeromecoutant,

Sorry, no update at this time. This isn't all that high on the priority list at the moment.

We are currently executing multiple parallel processes of Greentea at the moment instead of using the `--parallel` feature, as its a bit easier to manage lots of resources that way.

Please let us know if we can assist you in tracking down issues.

Thanks,
Brian

jeromecoutant, 2017-06-16:
Could you detail more "We are currently executing multiple parallel processes of Greentea"...?
Maybe this can be shared?
Thx

bridadan, 2017-06-16:
We mostly use it through Jenkins at the moment. However the equivalent would mean opening up multiple terminal windows and invoking `mbed test` or `mbedgt` in each one of those terminals.

jeromecoutant, 2018-06-26:
Hi
It seems that parallel tests are now working !

closed 2018-06-26 by jeromecoutant