ARMmbed/greentea #230: Tutorial or Example for Beginners to get started with mbedgt?

mbanders, 2017-07-03:
We have builds that compile an mbed app and flash it onto a K64F. That's all fine and works well for us, so we are not total beginners.

We would like to use `mbedgt` to do the most basic test: verify certain messages are printed during boot. If it sees those messages, and it doesn't see the string "error" or "fail", then pass the test. It's a very basic simple first test - and it's really all we're doing now as we develop an mbed app.

I've read the README, but it's still not clear how exactly we would do the above. Is there a nice example or tutorial somewhere? If not, maybe someone could suggest a starting point? I'd be willing to contribute a little tutorial - if I could figure out how to do this.

bridadan, 2017-07-05:
Hi @mbanders,

As part of greentea's test flow, it sends a synchronization message to the device. The device that runs the test binary will first wait for this message to be sent before sending the testing information to the host PC.

If you can't modify your binary to wait for this sync message, you may be interested in a functionality of greentea's subtool: htrun. It allows you to just check a device's log matches a template version of it: https://github.com/ARMmbed/htrun#testing-mbed-os-examples

mbanders, 2017-07-05:
> If you can't modify your binary to wait for this sync message ...

Are there clear instructions on how to do that somewhere? Although, I'm not sure our team will be keen on modifying our mbed app itself. I'd have to see the instructions. Without modification, are you saying our app won't be compatible for use with mbedgt?

It might take less time for us to just write a short python script using a serial library to read output from the device.

mbanders, 2017-07-05:
Also @bridadan where in the greattea README does it discuss the "synchronization message"? I can't find the word "synchronization" there.

bridadan, 2017-07-05:
@mbanders The synchronization (and in fact all communication with the device) is handled by the subtool I mentioned `htrun`. You can find information on the sync step here: https://github.com/ARMmbed/htrun#handshake

If you can't modify your application to wait for the sync message, then I'd recommend following the instructions in the link I posted above, it basically does what you're asking for if I understand correctly: https://github.com/ARMmbed/htrun#testing-mbed-os-examples

mbanders, 2017-07-13:
For any future person reading this thread, this is the command I needed to record serial output to a file:

```
mbedhtrun -p /dev/tty.usbmodem14412 --serial-output-file example.log --skip-flashing --baud-rate=115200
```

Which requires installing this first: https://github.com/ARMmbed/htrun

closed 2017-07-13 by mbanders