ARMmbed/greentea #231: Tests reported as skipped when they fail due to timeout

k-stachowiak, 2017-07-10:
When launching an mbedtls test suite via htrun a bug in one of the communication scripts leading the test data being sent incorrectly (the said bug is irrelevant to this report). This lead to the script fail on target, but from the host's point of view a timeout was observed. This was properly reported in the full log, but the test was reported as SKIPPED in the summary report.

In such case the system should report FAIL instead of SKIPPED.

bridadan, 2017-07-10:
Hi @k-stachowiak, can you post the full verbose (add `-v` to `mbed test` or `-V` if executing directly with `mbedgt`) test log here? Depending on how the test failed, it may report the other testcases as SKIPPED. However I would agree that if the test case that timedout should report FAIL/TIMEOUT.

k-stachowiak, 2017-07-11:
I included the most relevant log pieces. If it turns out not to be enough please let me know, I will strip it of any sensitive information it may contain and provide it. Below are the most of the details that I can give you quickly.

The behavior is triggered by intentionally corrupting the binary data transmission channel by mismatching the byte alignment. We investigated two cases: 4byte alignment on the host, 2bytes on the target, and 2bytes on the host, 4bytes on the target. I believe that the way we trigger the behavior, even though rather pathological and not likely in normal execution, is irrelevant. The issue isn't critical, but seems worth an investigation. Let me provide more details.

I am running the following command to run the tests:

    mbedgt -e . --test-spec testspec.json -g {device type}:module_name:10.2.123.43:3334 -v -V -c {device type} -p COM6

* When the host is 4 bytes and target 2 we observe the DISPATCH_INVALID_TEST_DATA error for several tests, which then are reported as skipped in the summary. For example:

```
[1499760004.70][HTST][INF] Running: Test bit set (Change existing value with a 1)
[1499760004.71][SERI][TXD] {{   H; I
S       49979687 I     I      I
S       66756903 I      }}
[1499760004.73][CONN][INF] found KV pair in stream: {{F;-4}}, queued...
[1499760004.74][HTST][INF] Error: DISPATCH_INVALID_TEST_DATA
```

And in the report:

    test case: 'Test bit set (Change existing value with a 1)' ................................... SKIPPED in 0.00 sec

* When the byte alignment is mismatched the other way round, i.e. host 2 bytes and target 4 bytes only one test runs and then the test system stops for few minutes and ends the tests and reports skip for all the other tests and timeout for the entire suite. It has been investigated that this case crashes the device, so this behavior is reasonable on the greentea side.

bridadan, 2017-07-11:
I believe I've addressed the issue in #232. If you have a chance, could you try it out and see if the behavior has been fixed?

k-stachowiak, 2017-07-12:
I tested with the most recent greentea repository and I observe the same behavior, e.g. when the test is executed:

```
[1499866662.22][HTST][INF] Running: X509 Get time (UTC invalid character in min)
[1499866662.23][SERI][TXD] {{   V;   E    SS      00113023u012Z E    +I       I       I       I       I       I       }}
[1499866662.25][CONN][INF] found KV pair in stream: {{F;-4}}, queued...
[1499866662.25][HTST][INF] Error: DISPATCH_INVALID_TEST_DATA
```
And then in the report we get:

```
        test case: 'X509 Get time (UTC invalid character in hour)' ................................... SKIPPED in 0.00 sec
```

It seems like the DISPATCH_INVALID_TEST_DATA is interpreted as a test skip.

Additionally, unfortunately I won't be able to provide any support regarding this issue for around 1.5 week. After that period I will gladly answer any further questions.

bridadan, 2017-07-12:
@k-stachowiak Thanks for continuing to look at this. Is the `DISPATCH_INVALID_TEST_DATA` part of your own testing? I don't think that's in the greentea or htrun code base.

k-stachowiak, 2017-07-12:
This is a symbol assigned to the error code in the mbedtls test scripts. The message is emitted here: https://github.com/ARMmbed/mbedtls/blob/ea67eeb4e8e6cdbaa8770935a848b11d231217bb/tests/suites/main_test.function#L462

bridadan, 2017-07-12:
@k-stachowiak Ah great thanks, I'll try to take a look at this later in the week.

theotherjimmy, 2017-09-20:
@bridadan Were you able to look into this? do you recall anything that might help me fix this bug?

bridadan, 2017-09-21:
@theotherjimmy Ah nope never did. @k-stachowiak Is this still issue?

k-stachowiak, 2017-09-21:
@bridadan We've been using the htrun script since my last message and this problem didn't occur again. I think it was specific to our environment.

theotherjimmy, 2017-09-21:
Huh. I'm going to close this, as we can no longer reproduce it.

closed 2017-09-21 by theotherjimmy