ARMmbed/greentea #236: mbedgt crash with float division by zero

jeromecoutant, 2017-09-21:
Hi

Here is my command:
mbedgt -V -v -t NUCLEO_F401RE-ARM,NUCLEO_F401RE-GCC_ARM,NUCLEO_F401RE-IAR,NUCLEO_F410RB-ARM,NUCLEO_F410RB-GCC_ARM,NUCLEO_F410RB-IAR,NUCLEO_F411RE-ARM,NUCLEO_F411RE-GCC_ARM,NUCLEO_F411RE-IAR --report-html=/c/xxx.html

It has crashed:

...
mbedgt: all tests finished!
mbedgt: shuffle seed: 0.3680156551
mbedgt: exporting to HTML file 
mbedgt: unexpected error:
	float division by zero
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedgt-script.py", line 11, in <module>
    load_entry_point('mbed-greentea==1.2.6', 'console_scripts', 'mbedgt')()
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 401, in main
    cli_ret = main_cli(opts, args)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_greentea_cli.py", line 1050, in main_cli
    html_report = exporter_html(test_report)
  File "c:\python27\lib\site-packages\mbed_greentea\mbed_report_api.py", line 747, in exporter_html
    int((test_results['single_test_passes']*100.0)/test_results['single_test_count']),
ZeroDivisionError: float division by zero

theotherjimmy, 2017-09-25:
Hi @jeromecoutant I'll get a PR up that fixes this pronto. Just a moment.

theotherjimmy, 2017-09-25:
@jeromecoutant Please take a look at #237. I would appreciate it if you could confirm that it fixes your problem.

jeromecoutant, 2017-09-25:
Thx.
I don't know if I can test non-released version of greentea.
I will try.

PR #237 (2017-09-25): Correct division by zero in report code
Resolves #236

Is breaking someone's setup. I'll make a release after this is merged
closed 2017-09-25 by theotherjimmy