ARMmbed/greentea #241: Very long test execution

jeromecoutant, 2017-09-29:
In my test bench, test execution is taking more and more time...
I wondering if you can help me to find the issue.

Here is an example of execution :

mbedgt: mbed-host-test-runner: started
[1506668998.88][HTST][INF] host test executor ver. 1.2.0
[1506668998.88][HTST][INF] copy image onto target...
[1506668998.89][COPY][INF] Waiting up to 60 sec for '07550221735D670B4C19F163' mount point (current is 'L:')...
        1 file(s) copied.
[1506669277.68][HTST][INF] starting host test process...
[1506669278.43][CONN][INF] starting connection process...
[1506669278.43][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1506669278.45][CONN][INF] initializing serial port listener...
[1506669278.45][PLGN][INF] Waiting up to 60 sec for '07550221735D670B4C19F163' serial port (current is 'COM7')...
[1506669278.45][HTST][INF] setting timeout to: 60 sec
[1506669323.86][SERI][INF] serial(port=COM7, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1506669323.86][SERI][INF] reset device using 'default' plugin...
[1506669324.13][SERI][INF] waiting 1.00 sec after reset
[1506669325.14][SERI][INF] wait for it...
[1506669325.17][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1506669325.17][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1506669325.17][CONN][INF] sending preamble '9bc110eb-92fd-4698-89b3-38d4502fe702'
[1506669325.22][SERI][TXD] {{__sync;9bc110eb-92fd-4698-89b3-38d4502fe702}}
[1506669325.28][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1506669325.33][CONN][INF] found SYNC in stream: {{__sync;9bc110eb-92fd-4698-89b3-38d4502fe702}} it is #0 sent, queued...
[1506669325.34][HTST][INF] sync KV found, uuid=9bc110eb-92fd-4698-89b3-38d4502fe702, timestamp=1506669325.327000
[1506669325.34][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1506669325.34][HTST][INF] DUT greentea-client version: 1.3.0
[1506669325.37][CONN][INF] found KV pair in stream: {{__timeout;30}}, queued...
[1506669325.39][HTST][INF] setting timeout to: 30 sec
[1506669325.40][CONN][INF] found KV pair in stream: {{__host_test_name;default_auto}}, queued...
[1506669325.42][HTST][INF] host test class: '<class 'mbed_host_tests.host_tests.default_auto.DefaultAuto'>'
[1506669325.42][HTST][INF] host test setup() call...
[1506669325.42][HTST][INF] CALLBACKs updated
[1506669325.42][HTST][INF] host test detected: default_auto

What I can see first is that file copy takes 279 seconds (1506668998.89 to 1506669277.68) !
Do you know what happened during this time ?

Thx

jeromecoutant, 2017-09-29:
Then, 45 seconds for serial init ?
[1506669278.45][HTST][INF] setting timeout to: 60 sec
[1506669323.86][SERI][INF] serial(port=COM7, baudrate=9600, read_timeout=0.01, write_timeout=5)

(it is a Windows 7 PC, 16 core, 16 GB RAM...)

theotherjimmy, 2017-09-29:
Hey @jeromecoutant,

I think you noticed that someone is looking at these issues now 😉.

Yeah, that seems wrong. What device are you using (I don't see that dev id in mbed ls)? Which test? also, this is probably an htrun problem.

jeromecoutant, 2017-09-29:
It is not related to the device, and same time is lost for all tests.
There are many devices connected to different USB hubs...

theotherjimmy, 2017-09-29:
@jeromecoutant It might be related to the fact that there are many devices.

jeromecoutant, 2017-09-29:
OS2 tests (via singletest.py) are very fast...

theotherjimmy, 2017-09-29:
Well, singletest.py does not use htrun

bridadan, 2017-09-29:
@jeromecoutant I can't say for sure, but mbed-ls slows down significantly (up to tens of seconds) if you have multiple devices plugged into a computer and being flashed. If boards are being flashed while you run greentea then this would probably explain the delay you're seeing.

The good news is @theotherjimmy is well underway refactoring mbed-ls to be wayyyy faster for CI. So this may not be a problem for too much longer.

theotherjimmy, 2017-09-29:
@bridadan `singletest.py` uses mbed-ls too, I think.

bridadan, 2017-09-29:
@theotherjimmy It depends how you use it. If you use the `--auto` flag then it uses mbed-ls, but you can also hard code the serial ports and mount points with JSON files.

jeromecoutant, 2017-09-29:
> singletest.py uses mbed-ls too, I think.

No, as mounted drive and com are configured in a json file.

theotherjimmy, 2017-09-29:
@jeromecoutant So it might be mbed-ls taking up all of the time here.

bridadan, 2017-09-29:
Going over the logs, it really seems like mbed-ls is the culprit here.

The line of interest is here: https://github.com/ARMmbed/htrun/blob/266d18dbccd1724e02e7e7be4bf2133f71cda953/mbed_host_tests/host_tests_conn_proxy/conn_primitive_serial.py#L47

This calls mbed-ls to make sure the board is attached correctly. You can see the next log print is here: https://github.com/ARMmbed/htrun/blob/266d18dbccd1724e02e7e7be4bf2133f71cda953/mbed_host_tests/host_tests_conn_proxy/conn_primitive_serial.py#L57

Which only comes after the big delay you're seeing.

So hopefully when the mbed-ls refactor is completed this won't be as much of an issue.

jeromecoutant, 2017-09-29:
And what about an option to force configuration to avoid mbedls multi call:
mbed test -i test_spec.json -M muts_all.json ?

ref: https://docs.mbed.com/docs/mbedmicro-api/en/latest/TESTING/#test-suite-core-singletestpy-script

theotherjimmy, 2017-09-29:
@jeromecoutant We can make such an option in greentea/htrun if you would like. I think it makes sense to have. Thoughts @bridadan ?

bridadan, 2017-09-29:
@jeromecoutant The mbedls calls in htrun are not because we don't know the port/mount point, its to make sure that the board has finished remounting after flashing. It's more complete and reliable then what singletest does (fixed timer wait then just continues the test).

It also deals with the case when testing under Linux where your mount point can change between flashes (depends on your automounter).

Now if you have complete control over your environment/mounting/etc (as you do generally in CI), then having this kind of feature may be a good idea. Either way, the mbedls case needs to happen. I'd recommend getting the mbedls fix in ASAP, as that will probably take care of most of the problems. We can come back to adding a forced configuration later.

jeromecoutant, 2017-10-03:
Hi
Please, make a new mbed-ls release version

theotherjimmy, 2017-10-03:
@jeromecoutant mbed-ls is not read for release just yet. I will be making a release, but it will be on the 1.2.x branch, and not contain any speedups like you want.

jeromecoutant, 2017-10-10:
Hi
Any update ? :-)

theotherjimmy, 2017-10-10:
Hey,

I have a bit more testing to get in, then I will be releasing mbed-ls 1.3.0, which should be a bit faster. In the mean time, the linux port is much faster than the windows one, so you could use that...

theotherjimmy, 2017-10-10:
armmbed/mbed-ls#231 is the last PR to merge before making the 1.3.0 release.

theotherjimmy, 2017-10-10:
@jeromecoutant  try https://github.com/ARMmbed/mbed-ls/releases/tag/v1.3.0

theotherjimmy, 2017-10-10:
There might be some speedups there.

jeromecoutant, 2017-10-10:
I am sorry, this doesn't work for my first try...
This looks very very verbose....

My command:
mbed test -t GCC_ARM -m DISCO_L072CZ_LRWAN1 -n tests-mbed_drivers-lp_timeout -v
==> mbedgt: no compatible devices detected

theotherjimmy, 2017-10-10:
@jeromecoutant what does mbedls -d give you?

jeromecoutant, 2017-10-11:
2 issues:
- all the boards from platform_database.py seem not to be listed by mbedls -l
- there is too much verbosity

theotherjimmy, 2017-10-11:
@jeromecoutant 

> all the boards from platform_database.py seem not to be listed by mbedls -l

That's a bug in mbed-ls, where it does not update the platform database after install

> there is too much verbosity

If you're talking about `-d` then just don't provide that switch on the command line. `-d` is for dumping enough information to make an automated test from, and not for an end user.

jeromecoutant, 2017-10-12:
> That's a bug in mbed-ls, where it does not update the platform database after install

OK, I have seen https://github.com/ARMmbed/mbed-ls/issues/232

> If you're talking about -d then just don't provide that switch on the command line. -d is for dumping enough information to make an automated test from, and not for an end user.

Yes, of course.
v1.3.1 looks better, but there are still debug prints:

ex from a test log :

[1507809517.94][COPY][INF] Waiting up to 60 sec for '07950221014C67123E08F1BA' mount point (current is 'D:')...
[1507809517.98][mbedls.lstools_base]Call to deprecated function list_mbeds_by_targetid. List formatting methods are deprecated to simplify the API. Please use 'list_mbeds' instead.
[1507809517.98][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1507809518.07][root]Found htm target id, 07950221014C67123E08F1BA, for usb target id 066DFF535550755187012736
[1507809518.08][mbedls.lstools_base]MBED with target id '066EFF525150677267112956' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1507809518.08][mbedls.lstools_base]MBED with target id '066AFF485550755187193041' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1507809518.08][mbedls.lstools_base]MBED with target id '066EFF485550755187052716' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1507809518.08][mbedls.lstools_base]MBED with target id '066BFF545253675187212436' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1507809518.08][mbedls.lstools_base]MBED with target id '066EFF323535474B43122727' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1507809518.08][mbedls.lstools_base]MBED with target id '066BFF555352845187102948' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1507809518.08][mbedls.lstools_base]MBED with target id '066DFF515456707067123833' is connected, but not mounted. Use the '-u' flag to include it in the list.
[1507809518.08][mbedls.lstools_base]MBED with target id '066CFF535752877167012638' is connected, but not mounted. Use the '-u' flag to include it in the list.

This block is printed 8 times for 1 test!
(my command : mbed test -t GCC_ARM -m DISCO_F429ZI -n tests-mbed_drivers-lp_timeout -v)

theotherjimmy, 2017-10-12:
> This block is printed 8 times for 1 test!

It seems like htrun/greentea are not caching intermediate results of `list_mbeds` correctly. This would certainly lead to the behavior you described: it would be really slow.

jeromecoutant, 2017-10-13:
Hi
I am quite happy with the last mbedls release !
We can close this issue, 
or you prefer to keep it alive in order to implement IOTMORF-1380 ?

theotherjimmy, 2017-10-13:
> or you prefer to keep it alive in order to implement IOTMORF-1380 ?

Nope!

closed 2017-10-13 by theotherjimmy